prusa3d/PrusaSlicer #13672: Crashing about 90% of the time while exporting G-Code

archonik, 2024-11-27:
### Description of the bug

Hello,

anytime i try to export g-code, there is about 90% chance that prusaslicer will crash. No error, nothing.

I can import models, slice the models just fine. When i click on export g-code, thats when it crashes. 

Was also happening (much less frequently) on 2.7.4.

100% not due to lack of resources.

My PC Specs:
I7-12700KF (all other apps are stable)
96 GB Ram (tested with memtest)
15TB Total storage

### Project file & How to reproduce

- Import 2 parts of model
- Slice
- Export g-code

[BracketFix - Part 1.zip](https://github.com/user-attachments/files/17933643/BracketFix.-.Part.1.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.1+ win64

### Operating system

Windows 11 - 23H2 - 22631.4460

### Printer model

I3-MK3S+

u89djt, 2024-11-27:
(fellow user following the hub for useful intel(pun intended))
I opened that file, sliced and exported 20 times and got no crash. I happen to be running ryzens in everything.
I see from googling that people have had trouble with the e cores of your processor, and there are bios updates and workarounds available.

archonik, 2024-12-02:
So i narrowed down the problem. The problem is caused when trying to export directly to network drive. 

I have a samba share setup to my octoprint server directly. When trying to export direclty on local HDD, everything is okay. When trying to export directly to network drive, thats when it crashes. Network drive otherwise works just fine, i can copy/create/delete files normally.

u89djt, 2024-12-02:
Aye, "otherwise works fine" is where the wild things are :) You've applied all the fixes for CPU/mobo and such?

u89djt, 2024-12-02:
You've got computer scientists and other IT professionals out here gritting their teeth. Whatever causes the problem is still there waiting to bite you in another scenario.

closed 2026-09-29 by prusaslicer-bot[bot]