prusa3d/PrusaSlicer #13676: Crash while slicing - XL 5H with imported 3mf

blillesand, 2024-11-27:
### Description of the bug

Imported 3mf file that was purchased from Cobotech via Cults 3D.  Prusaslicer 2.9.0 crashes when slicing this file (works fine on 2.8)

### Project file & How to reproduce

1) Import 3mf file (Cobotech Articulated Turkey Chef, purchased on Cults 3d)
2) Slice

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.9.0-alpha1-win64-202411202012

### Operating system

Windows 11

### Printer model

Prusa XL 5H

Jan-Soustruznik, 2024-11-28:
Hello @blillesand,Could you please share your 3MF file so we can investigate the crash? Thank you, and have a great day!

BillC58, 2024-12-08:
I have this crash too. Here is my .3MF file.

[2024 Reindeer XL trees.zip](https://github.com/user-attachments/files/18053178/2024.Reindeer.XL.trees.zip)

themanyone, 2024-12-09:
> I have this crash too. Here is my .3MF file.
> 
> [2024 Reindeer XL trees.zip](https://github.com/user-attachments/files/18053178/2024.Reindeer.XL.trees.zip)

Testing with [debug] semver.c 2.9.0-alpha1 = 2.9.0, alpha1 (null) (git) shows some issue with colored_polygons being empty.
```
prusa-slicer: /home/k/Downloads/src/Prusa/src/libslic3r/MultiMaterialSegmentation.cpp:1809: Slic3r::segmentation_by_painting(const PrintObject&, const std::function<ModelVolumeFacetsInfo(const ModelVolume&)>&, size_t, float, float, IncludeTopAndBottomLayers, const std::function<void()>&)::<lambda(const tbb::detail::d1::blocked_range<long unsigned int>&)>: Assertion `!colored_polygons.empty()' failed.
```
If all you need is the model, the .3mf file may be opened with GUI programs like Ark and File Archiver. From there, you can export the model sans configurations.

closed 2024-12-20 by lukasmatena