prusa3d/PrusaSlicer #13678: Weird geometry generated on negative seam gap value, worse with scarf joint

gaiuscosades, 2024-11-27:
### Description of the bug

Non existent geometry is generated that seems to be determined on a negative seam gap value.
Enabling Scarf joint results in additional bad geometry.
[slicer_negative_seam_gap_bug.zip](https://g
[image]
ithub.com/user-attachments/files/17940353/slicer_negative_seam_gap_bug.zip)

### Project file & How to reproduce

Open project
Look at geometry
Comapre with sliced geometry (see screenshot)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.9.0-alpha1

### Operating system

Windows 11

### Printer model

Prusa Mini+

neophyl, 2024-11-28:
Could you edit your post to fix the project link.  Your image seems to have cut it in 2.

gaiuscosades, 2024-11-28:
done.

Thanks, for noticing, just learned that drag-drop does not append at the end...

gaiuscosades, 2024-11-28:
Just noticed that this behavior seems to completely disappear when arc-welding is disabled.

So this seems to be a bug that arises when loops are more than completely closed (due to negative seam gap value) and then arc welder not knowing what to do with the result.

neophyl, 2024-11-28:
Nicely spotted :)  I was monitoring the issue as I've been using a negative seam gap on my Neptune 3 Max and was having no problems, its helped tremendously, far more than scarf seams in fact.  So was curious as to potential issues.

gaiuscosades, 2024-11-28:
Cool! I also wanted to fine tune that value, as I think it has lots of potential in improving seam quality.

Can't share the project, but for me the described bug first resulted in something like this on my Prusa XL. Yikes ^^
[image]

TerPuc, 2024-11-28:
Dear @gaiuscosades, thanks for the report! We'll look at that.

Issue created: SPE 2597

FidelCapo, 2024-12-16:
This issue should be fixed in version 2.9.0-beta1.
Closing.

closed 2024-12-16 by FidelCapo