prusa3d/PrusaSlicer #13681: Arrange command should account for supports "first layer expansion" setting

cgri, 2024-11-28:
**Environment**
Prusa Slicer version: 2.8.1
Operating system: Windows 11

**Steps to reproduce**
* Select MK4S 0.2 mm speed profile
* Add two parts that require supports on their edges or corners.
* Enable supports
* Use auto-arrange
* Slice

**Expected behavior**
The auto-arrange command should arrange the items in a way that the support structures do not overlap (at least with default prusa printer profiles).

**Actual behavior**
When using the auto-arrange command with supports enabled, it arranges items so close to each other that the default "first layer expansion" setting in the MK4S default printer profiles sometimes causes overlapping support structures. The slicer then warns about overlapping items.

**Possible solutions**
Either the default for the "first layer expansion" should be reduced to 1.5 Millimeters in the prusa printer profiles (simple solution) or the auto-arrange should account for that (maybe not so simple solution).

**Screenhot**
The following screenshot shows that the support expansion of the top left and bottom left parts overlap. The overlap might be negligible in this case, but has been more in other cases.
[image]

Not a big deal, would just be a nice improvement.

Anyway, thanks a lot for your great work!

u89djt, 2024-11-28:
(fellow user following the hub) You can right-click the arrange icon and set the spacing and repositioning algorithm you want.

u89djt, 2024-11-28:
Following your instructions with everything set to defaults, including the arrange parameters, doesn't lead to overlaps, so you've changed some of your settings. The arrange parameters persist between sessions, so perhaps you modified those? Post the project - you need to zip the 3mf file - and it should be possible to work out what's going on.
[Shape-Box.zip](https://github.com/user-attachments/files/17949822/Shape-Box.zip)
[image]

u89djt, 2024-11-28:
Following your instructions with defaults for everything including the layout dialog doesn't lead to overlap. If you can post the 3mf file (you need to zip it to post) it should be possible to work out what settings you've changed.
[Shape-Box.zip](https://github.com/user-attachments/files/17949924/Shape-Box.zip)
[image]

u89djt, 2024-11-28:
Ah, if I manually rotate the two on the right, we get an overlap.
[manualShape-Box.zip](https://github.com/user-attachments/files/17949968/manualShape-Box.zip)
[image]

cgri, 2024-12-03:
> (fellow user following the hub) You can right-click the arrange icon and set the spacing and repositioning algorithm you want. It doesn't solve everything, but you have some opportunities there.

Oh, I actually didn't know this. 🙈 I think this pretty much resolves this issue.

Thanks a lot for your help!

closed 2024-12-03 by cgri