prusa3d/PrusaSlicer #13682: Filament types are not available in Configuration Wizard

777Samael, 2024-11-28:
### Description of the bug

The setup wizard for the MK4S MMU3 printer lists only some compatible filament types, only PLA, PETG, and a few others are available, but ABS, ASA, FLEX, PC, and others are not.

### Project file & How to reproduce

[leg-R4.zip](https://github.com/user-attachments/files/17949981/leg-R4.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.1+win64

### Operating system

Win10

### Printer model

MK4S MMU3

777Samael, 2024-11-28:
[image]

rtyr, 2024-11-28:
That is expected behavior, those are currently supported materials for MMU prints. Enable and use classic non-mmu MK4S printer profile for single material prints.

closed 2024-11-28 by rtyr

777Samael, 2024-11-28:
Srsly? I understand FLEX, but BVOH, ABS, ASA, HIPS, PA, PVA, PCTG, PP, and PC can be printed using MMU3. I've tested this, and you have tested it, and it works. If you need more people, just let me know. I know manual and automated testing workflows. I can help you with tests even remotely on my printer.

TheNore, 2024-12-16:
Seconding @777Samael on this one. I have been able to print these materials on the MMU for years before this. Why now are they "not supported" At least give a work around with a disclaimer for the end user.

paulrobello, 2025-05-15:
I third this, I want to start printing with PCTG and have the MMU3, So now I have to modify the default PETG profile with all the PCTG settings to create a custom profile. Not a big deal but a simple tweak to the filament import wizard would be a big win for the user experience.

andyman1222, 2026-11-12:
This issue needs to be reopened.

I can apparently select PVA in single material mode without the MMU3... But add the MMU3 and it's not supported. What is even the point of printing a soluble material in single material mode?