prusa3d/PrusaSlicer #13684: 2.9.0 Alpha 1 removes wipe_tower_x and wipe_tower_y from place holders

joseph-greiner, 2024-11-28:
### Description of the bug

Custom gcode for tool change errors because wipe_tower_x and wipe_tower_y were removed from placeholders.

### Project file & How to reproduce

under custom gcode add:
G0 X{wipe_tower_x} Y{wipe_tower_y} F{travel_speed*60}

Slicing will error because the place holders no longer exist.

[image]

If the place holders will not be returning, how can one move the tool head to the wipe tower before calling the tool change gcode?

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.9.0 Alpha 1

### Operating system

All

### Printer model

All

lukasmatena, 2024-12-22:
Thanks. We are aware of the problem. It will be addressed in the next version.

scetron, 2025-01-18:
I also have the same issue. Thank you for all the hard work!

lukasmatena, 2025-02-18:
Should be fixed in [PrusaSlicer 2.9.1-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.1-alpha1). Closing.