prusa3d/PrusaSlicer #13685: Unable to Remove Support-like Features in Infill

TheOneTrueRazor, 2024-11-29:
### Description of the bug

I’m currently printing 12mm tall Christmas ornaments on my 5-tool XL. Layer height is .2mm, infill is set to 10%. I use single color/tool up to the 10.6mm height. Then PrusaSlicer begins to use the other tools for the multimaterial painting features.

Throughout the slice, multiple “towers” are made in the infill. My assumption is that they are there to support the details (nozzle paths) on the top layer. 

This behavior appears on multiple projects similar in design.

### Project file & How to reproduce

In an effort to remove them, I attempted to print a solid layer just before the multimaterial painting. That did not solve my problem, in fact even with a 100% model infill, these features continue to exist.

While I understand the need for them with lower infill, I expect their numbers to decrease as the infill density is increased and completely disappear when the infill is 100%. They increase print time, add complexity to the print, and cause top and bottom surface imperfections. 
[Archive.zip](https://github.com/user-attachments/files/17961080/Archive.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.1

### Operating system

MacOS 12.7.6

### Printer model

XL 5-Tool

u89djt, 2024-11-30:
https://github.com/prusa3d/PrusaSlicer/issues/13685
(Fellow user following the hub)
TLDR: Lots of mesh bodies with mismatches where they meet at the curvy svg lines. Not a bug.
Taking the svg you posted here
https://forum.prusa3d.com/forum/prusaslicer/svg-multi-material-painting-help/?language=pl
And building in Fusion 360 with added outlines and a yellow border to mimic your project model, I get the following for comparison. (Plus an appreciation of how much work you put into it.)
Your original is bottom left, and a copy of it with the painting deleted is above. My fan art is on the right, and then with all colours set to grey above it.
[rebuilt Hornets.zip](https://github.com/user-attachments/files/17967215/rebuilt.Hornets.zip)
[image]
My model is split into part groups for each colour so it can have a definitely solid base.
We also see the effects of mesh surface mismatches in the topmost array of colour parts in the all-grey version of mine.
[image]
But not in the main solid body of the base. The view is of the layer where the bridge infill for the void perimeters are placed.
[image]
It seems like the problem is that the approximation introduced by meshing causes voids between bodies that the slicer doesn't extrude through.
You can juuuuust see some of the near-misses:
[image]
It would be great if it could be solved automagically, but people build tight non-zero tolerances into their models that the slicer has to also respect, so perhaps that doesn't have a tractable answer.
It should be possible to build perfect vertex coincidence between objects either during the build, or maybe with a Meshlab filter. I might have a go at both. The right solution could potentially be to rebuild the svg so that the native import works well.
Oh. The way to do it will be to build vertical mesh surfaces for the lines in the svg, duplicate where objects meet and cap them off. Maybe tomorrow.

u89djt, 2024-12-01:
This is the cleanest version of your original mesh (starting with that, not the svg) that I can make by sequentially subtracting faceted parts from a merged whole. It still glitches in two locations if it's loaded as a step file (an ensemble of abutting faceted bodies). Instead of continuous towers, there are islands above each other that don't appear at the top surface. 
[hornets clean.zip](https://github.com/user-attachments/files/17970496/hornets.clean.zip)
[image]
If you set the slice gap closing radius to 0.005 the glitches in this project disappear. Not in your original project, sadly.
I've run out of ideas for the moment.

TheOneTrueRazor, 2024-12-02:
Wow, thank you so much for breaking this down for me. I think I understand most of what you are saying, but will have to play around in Fusion a little more. The original logo file I had to work from was very intimidating. I don't have a lot of experience with Adobe (I'm guessing this is what they used) and Fusion 360. To be honest, I feel like a bull in a China shop that somehow managed to get from one side to the other without breaking EVERYTHING. LOL

I can't express how grateful I am for you taking the time to do this and explain it like you did.

u89djt, 2024-12-02:
That's great! I learned a bunch of stuff that will probably come in useful while doing it. My fusion file for it is hot garbage because a lot of finding out had to happen. So there are two angles: what should the developers care about here, and where should we take the development of the object. What's the licensing around the svg file? I could potentially post a printables object and neaten it up.

TheOneTrueRazor, 2024-12-03:
This was the right answer! I was able to go back in and clean it up myself in Fusion 360. I created a base as you did and then went and grouped my color objects as you did. Making that change cutoff about an hour per print! I have similar style of ornaments for other teams my kids play on (yes we're very active) and was able to reduce print times similarly by ~ 33%. I do developer work as a profession. Like programming, there are many different ways to accomplish the same task. It's up to us to find what is the most efficient. :-)

I'd rather not post it on Printables. I don't want to upset someone if they see it over there. As I get better with this I have a few ideas now for future projects. I'll put some of that stuff up. Thanks again. I hope I can return the favor some day.

u89djt, 2024-12-03:
Just do it for the next person you bump into, and they might be inspired to do the same :)

closed 2025-01-09 by Razor412