prusa3d/PrusaSlicer #13687: Crash after opening Physical Printer Tab

Flo082002, 2024-11-29:
### Description of the bug

Crash after opening Physical Printer Tab

https://github.com/user-attachments/assets/df8325c9-f436-46cf-bb7e-23b9fd03e7b4

### Project file & How to reproduce

[Unbenannt.zip](https://github.com/user-attachments/files/17962294/Unbenannt.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.9.0-alpha1+flathub.org

### Operating system

Ununtu 24.04

### Printer model

Anycubic i3 Mega Klippter

Noisyfox, 2024-12-07:
https://github.com/SoftFever/OrcaSlicer/issues/7210#issuecomment-2524739398

themanyone, 2024-12-08:
Seems to be fixed in git. There were also some layer height assertions.
Variable layer height causes assertion->crash when switching printers.
See if #13691 fixes it.

closed 2026-09-29 by prusaslicer-bot[bot]