prusa3d/PrusaSlicer #13689: new version . filler leaves incomplete pattern

selyar, 2024-11-30:
### Description of the bug

[image]
[image]
new version . filler leaves incomplete pattern

### Project file & How to reproduce

[Unnamed-Body.zip](https://github.com/user-attachments/files/17963868/Unnamed-Body.zip)

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Sürüm 2.9.0-alpha1-win64

### Operating system

windows7 

### Printer model

artilery x1

rtyr, 2024-12-04:
Could you please provide your 3MF project file?

rtyr, 2024-12-04:
never mind, we were able to reproduce it

closed 2026-09-29 by prusaslicer-bot[bot]