prusa3d/PrusaSlicer #13690: Feature Request: Keep Max in Variable layer height

DavidBjerreBjoerklund, 2024-11-30:
**Is your feature request related to a problem? Please describe.**
When doing detailed models I have already said the maximum desired layer height. Some areas would be nice to make even finer and sometimes a little help is... well helpfull, when detecting those areas.

I would like a "keep Max" added to variable layer height as well as the keep min.
[image]

neophyl, 2024-11-30:
The maximum and minimum heights for VLH are bound by the settings for Printers>Extruder>Layer Height Limits.  They are global ones which determine what you can set on your print profiles too.  So if you wanted so limit VLH to say 0.25 then just set the Max in the profile to 0.25.

themanyone, 2024-12-09:
There are problem with setting max in the profile to 0.25 and using printer profiles to set VLH limits.

It makes Prusaslicer assert and exit to console when switching printers (to ones that don't support the variable layer heights sliced previously). Pull request #13691

closed 2026-09-29 by prusaslicer-bot[bot]