prusa3d/PrusaSlicer #13694: Linux 2.9.0-alpha1 - compillation failed: libslic3r-arrange.so linker relocation error R_X86_64_TPOFF32

mkyral, 2024-11-30:
### Description of the bug

I'm trying to compile latest 2.9.0-alpha1 on my Gentoo and it fails on linker relocation error:

```
[345/514] : && /usr/bin/x86_64-pc-linux-gnu-g++ -fPIC -march=znver3 -mshstk --param=l1-cache-line-size=64 --param=l1-cache-size=32 --param=l2-cache-size=512 -O2 -pipe -fext-numeric-literals -Wall -Wno-reorder  -Wl,-O1 -Wl,--as-needed -Wl,-z,pack-relative-relocs -shared -Wl,-soname,libslic3r-arrange.so -o src/slic3r-arrange/libslic3r-arrange.so src/slic3r-arrange/CMakeFiles/slic3r-arrange.dir/src/Beds.cpp.o src/slic3r-arrange/CMakeFiles/slic3r-arrange.dir/src/NFP/NFP.cpp.o src/slic3r-arrange/CMakeFiles/slic3r-arrange.dir/src/NFP/NFPConcave_Tesselate.cpp.o src/slic3r-arrange/CMakeFiles/slic3r-arrange.dir/src/NFP/EdgeCache.cpp.o  src/libslic3r/liblibslic3r.a  src/libslic3r/liblibslic3r.a  /usr/lib64/libboost_system.so.1.85.0  /usr/lib64/libboost_log.so.1.85.0  /usr/lib64/libboost_filesystem.so.1.85.0  /usr/lib64/libboost_locale.so.1.85.0  /usr/lib64/libboost_thread.so.1.85.0  /usr/lib64/libboost_regex.so.1.85.0  /usr/lib64/libboost_chrono.so.1.85.0  /usr/lib64/libboost_atomic.so.1.85.0  /usr/lib64/libboost_date_time.so.1.85.0  /usr/lib64/libboost_nowide.so.1.85.0  src/clipper/libclipper.a  /usr/lib64/libexpat.so.1.10.0  bundled_deps/glu-libtess/libglu-libtess.a  /usr/lib64/libqhullcpp.a  /usr/lib64/libqhull_r.so.8.0.2  src/libslic3r/liblibslic3r_cgal.a  /usr/lib64/libgmpxx.so  /usr/lib64/libmpfr.so  /usr/lib64/libgmp.so  -ldl  /usr/lib64/libpng.so  /usr/lib64/libjpeg.so  bundled_deps/libqoi.a  bundled_deps/libsemver.a  bundled_deps/admesh/libadmesh.a  bundled_deps/liblocalesutils.a  /usr/local/lib64/libbgcode_convert.a  /usr/local/lib64/libbgcode_binarize.a  /usr/local/lib/libheatshrink_dynalloc.a  /usr/local/lib64/libbgcode_core.a  bundled_deps/miniz/libminiz_static.a  /usr/lib64/libopenvdb.so  /usr/lib64/libz.so  /usr/lib64/libboost_iostreams.so.1.85.0  /usr/lib64/libImath-3_1.so.29.10.0  -lm  /usr/lib64/libtbb.so.12.9  /usr/lib64/libtbbmalloc.so.2.9  /usr/lib64/libnlopt.so && :
FAILED: src/slic3r-arrange/libslic3r-arrange.so
: && /usr/bin/x86_64-pc-linux-gnu-g++ -fPIC -march=znver3 -mshstk --param=l1-cache-line-size=64 --param=l1-cache-size=32 --param=l2-cache-size=512 -O2 -pipe -fext-numeric-literals -Wall -Wno-reorder  -Wl,-O1 -Wl,--as-needed -Wl,-z,pack-relative-relocs -shared -Wl,-soname,libslic3r-arrange.so -o src/slic3r-arrange/libslic3r-arrange.so src/slic3r-arrange/CMakeFiles/slic3r-arrange.dir/src/Beds.cpp.o src/slic3r-arrange/CMakeFiles/slic3r-arrange.dir/src/NFP/NFP.cpp.o src/slic3r-arrange/CMakeFiles/slic3r-arrange.dir/src/NFP/NFPConcave_Tesselate.cpp.o src/slic3r-arrange/CMakeFiles/slic3r-arrange.dir/src/NFP/EdgeCache.cpp.o  src/libslic3r/liblibslic3r.a  src/libslic3r/liblibslic3r.a  /usr/lib64/libboost_system.so.1.85.0  /usr/lib64/libboost_log.so.1.85.0  /usr/lib64/libboost_filesystem.so.1.85.0  /usr/lib64/libboost_locale.so.1.85.0  /usr/lib64/libboost_thread.so.1.85.0  /usr/lib64/libboost_regex.so.1.85.0  /usr/lib64/libboost_chrono.so.1.85.0  /usr/lib64/libboost_atomic.so.1.85.0  /usr/lib64/libboost_date_time.so.1.85.0  /usr/lib64/libboost_nowide.so.1.85.0  src/clipper/libclipper.a  /usr/lib64/libexpat.so.1.10.0  bundled_deps/glu-libtess/libglu-libtess.a  /usr/lib64/libqhullcpp.a  /usr/lib64/libqhull_r.so.8.0.2  src/libslic3r/liblibslic3r_cgal.a  /usr/lib64/libgmpxx.so  /usr/lib64/libmpfr.so  /usr/lib64/libgmp.so  -ldl  /usr/lib64/libpng.so  /usr/lib64/libjpeg.so  bundled_deps/libqoi.a  bundled_deps/libsemver.a  bundled_deps/admesh/libadmesh.a  bundled_deps/liblocalesutils.a  /usr/local/lib64/libbgcode_convert.a  /usr/local/lib64/libbgcode_binarize.a  /usr/local/lib/libheatshrink_dynalloc.a  /usr/local/lib64/libbgcode_core.a  bundled_deps/miniz/libminiz_static.a  /usr/lib64/libopenvdb.so  /usr/lib64/libz.so  /usr/lib64/libboost_iostreams.so.1.85.0  /usr/lib64/libImath-3_1.so.29.10.0  -lm  /usr/lib64/libtbb.so.12.9  /usr/lib64/libtbbmalloc.so.2.9  /usr/lib64/libnlopt.so && :
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: src/libslic3r/liblibslic3r_cgal.a(VoronoiUtilsCgal.cpp.o): relocation R_X86_64_TPOFF32 against symbol `_ZGVZN4CGAL4LazyINS_7Point_2INS_16Simple_cartesianINS_11Interval_ntILb0EEEEEEENS1_INS2_I10__gmp_exprIA1_12__mpq_structS9_EEEEENS_19Cartesian_converterISB_S5_NS_12NT_converterISA_S4_EEEEE4zeroEvE1z' can not be used when making a shared object; recompile with -fPIC
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: failed to set dynamic section sizes: bad value
collect2: error: ld returned 1 exit status
```

### Project file & How to reproduce

I bump version in 2.8.1 ebuild and run emerge.

**Configure options**

```
-- The C compiler identification is GNU 13.3.1
-- The CXX compiler identification is GNU 13.3.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/x86_64-pc-linux-gnu-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/x86_64-pc-linux-gnu-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- SLIC3R_ASAN: OFF
-- SLIC3R_BUILD_SANDBOXES: OFF
-- SLIC3R_BUILD_TESTS: no
-- SLIC3R_ENABLE_FORMAT_STEP: ON
-- SLIC3R_FHS: ON
-- SLIC3R_GTK: 3
-- SLIC3R_GUI: ON
-- SLIC3R_LOG_TO_FILE: OFF
-- SLIC3R_MSVC_COMPILE_PARALLEL: ON
-- SLIC3R_MSVC_PDB: ON
-- SLIC3R_OPENGL_ES: OFF
-- SLIC3R_PCH: OFF
-- SLIC3R_REPO_URL: OFF
-- SLIC3R_STATIC: OFF
-- SLIC3R_UBSAN: OFF
-- SLIC3R_WX_STABLE: ON
-- SLIC3R_STATIC: OFF
-- CMAKE_PREFIX_PATH: (default)
-- Found PkgConfig: /usr/bin/x86_64-pc-linux-gnu-pkg-config (found version "2.3.0")
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- Found DBus: /usr/include/dbus-1.0;/usr/lib64/dbus-1.0/include
-- Found Boost: /usr/lib64/cmake/Boost-1.85.0/BoostConfig.cmake (found suitable version "1.85.0", minimum required is "1.66.0") found components: system filesystem thread log locale regex chrono atomic date_time iostreams nowide
-- Found Eigen3: /usr/include/eigen3 (Required is at least version "3.3.7")
-- Boost::boost exists
-- TBB found in /usr/lib64/cmake/TBB
-- Could NOT find CURL (missing: CURL_DIR)
-- Falling back to MODULE search for CURL...
-- Found CURL: /usr/lib64/libcurl.so (found version "8.10.1")
-- Found ZLIB: /usr/lib64/libz.so (found version "1.3.1")
-- EXPAT found in /usr/lib64/cmake/expat-2.6.4
-- Found PNG: /usr/lib64/libpng.so (found version "1.6.44")
-- Found OpenGL: /usr/lib64/libOpenGL.so
-- Could NOT find GLEW (missing: GLEW_DIR)
-- Falling back to MODULE search for GLEW...
-- Found GLEW: /usr/include (found version "2.2.0")
-- Found NLopt in '/usr/lib64'.
[... 33 lines cut ...]
Compiler flags:
C               -march=znver3 -mshstk --param=l1-cache-line-size=64 --param=l1-cache-size=32 --param=l2-cache-size=512 -O2 -pipe
C++             -march=znver3 -mshstk --param=l1-cache-line-size=64 --param=l1-cache-size=32 --param=l2-cache-size=512 -O2 -pipe -fext-numeric-literals -Wall -Wno-reorder
Linker flags:
Executable      -Wl,-O1 -Wl,--as-needed -Wl,-z,pack-relative-relocs
Module          -Wl,-O1 -Wl,--as-needed -Wl,-z,pack-relative-relocs
Shared          -Wl,-O1 -Wl,--as-needed -Wl,-z,pack-relative-relocs
```

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0_alpha1

### Operating system

Gentoo linux

### Printer model

Prusa MK4

fordfrog, 2024-12-19:
we have prusaslicer-9999 which applies fpic patch for relevants parts of the code. but that alpha does not run as some libraries are not installed. and the rc is broken too now as we have boost-1.87 with which none of prusaslicer versions we have builds without issues.

cryptomilk, 2024-12-20:
I really wonder why PrusaSlicer doesn't set `set(CMAKE_POSITION_INDEPENDENT_CODE ON)` in `src/CMakeLists.txt`.

cryptomilk, 2024-12-20:
Just setting it on the slic3r-arrange target will lead to a segfault in the slic3rutils test. I think we really want this turned on for all libraries and binaries.

fordfrog, 2024-12-21:
i get the same sigsegv even when setting fpic globally

cryptomilk, 2024-12-21:
The segfault is something else.

adelton, 2025-01-02:
I assume

```diff
     -DPrusaSlicer_BUILD_DEPS:BOOL=ON \
     -DOPENVDB_USE_STATIC_LIBS=1 \
     -DCMAKE_EXE_LINKER_FLAGS=-lcrypto \
+    -DCMAKE_POSITION_INDEPENDENT_CODE=ON \
 %if %{with perltests}
     -DSLIC3R_PERL_XS=1
 %endif
```

in a `.spec` file would be equivalent, right? At least it compiles for me.

For the record (note to self), the segfault is discussed in https://github.com/prusa3d/PrusaSlicer/issues/13813.

cryptomilk, 2025-01-02:
Yes, this is the equivalent.

adelton, 2025-01-04:
The https://github.com/prusa3d/PrusaSlicer/pull/13896 seems to fix the issue (at least on Fedora rawhide and 41) while not introducing the segfault situation.

closed 2026-09-29 by prusaslicer-bot[bot]