prusa3d/PrusaSlicer #13696: Filament extrusion multiplier not working.

klinwer, 2024-11-30:
### Description of the bug

Changing the value of filament extrusion multiplier has no effect to the print result. The same issue with PrusaSlicer 2.8.1.

### Project file & How to reproduce

Changing the filament extrusion multiplier value from 0,8 to 0,7 as example has no effect to the print.

Also reported:
BUG Extrusion multipler set in filament tab is not reflected in gcode #9244
Flow Extrusion multiplier ignored #8190
Extrusion Multiplier not working? #2648

[cube-40-40-40_08.zip](https://github.com/user-attachments/files/17966006/cube-40-40-40_08.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.9.0-alpha1-win64-202411202012

### Operating system

Windows 11, Version 10.0.22631 Build 22631

### Printer model

MK4S

u89djt, 2024-12-02:
Compare these extrusion quantities in your project  for 0.8 and 0.7 below. They're working as intended.
[image]
[image]
If you go back the the issues you linked to, you'll find that this is explained.
All I can suggest as a fellow user who has also got lost in the settings on occastion, is that you might have modified settings for a filament other than the one you're printing with.

klinwer, 2024-12-03:
Thank you for the fast response. I checked the multiplier value and can confirm that it is working. I am still lost with this issue since my upgrade from MK4 to MK4S.
This was a working filament, now I get this. The multiplier value of 0.8 is only for testing purpose.

[image]

u89djt, 2024-12-03:
I was too fast - I should have indicated that I'm a fellow user, not a dev.
Very happy to continue the search, as long as you're happy for me to get things wrong with you.
That is very obviously massively under-extruded. The paths look good.
If you're extruding 0.8 of the filament you need to extrude, then I think that's what you might see.
When you say 0.8 is for testing purposes, what are you actually trying to achieve?
If you want narrower extrusions, you need to do set the extrusion width here:
[image]
When you say you upgraded, did you buy it built or as a kit? If it was as a kit, is it possible that the extruder control didn't go together well? Transport looks good, but you're obviously getting the plastic at a fairly constant fraction of what it should be.
There's a lot to unpack here, so stick with it.

klinwer, 2024-12-03:
This is a kit upgrade, the second one. I started with an MK3. The upgrade went well, also the self test. I did'nt disassemble the extruder. The problem are the drops on the nozzle. Thats why I try to reduce the extrusion value.

u89djt, 2024-12-03:
OK, so there's a long-ish road ahead. This definitely isn't a github issue, so go ahead and close it.
The drips are generally referred to as "oozing". They're due to the filament absorbing atmospheric moisture and expanding as that moisture is boiled by the heat.
The correct solution is to dry the filament, which involves heating it to maybe 50 degrees C for a few hours, then storing it in an airtight container with desiccant.
When you've dried the filament as much as you're going to, oozing is then dealt with by retraction of the filament while not printing. There's a certain amount of that built into filament profiles. Search for "retract" in the box at the top of the slicer.
If you reduce the extrusion, your prints will generally not work, and if they do, they'll look terrible. You'll get gaps all over the place as the filament sticks to itself and pulls into lumps just like you're seeing on the top of your cube.
Ask any questions and say anything that seems relevant to you.

u89djt, 2024-12-03:
You could think about moving the discussion to the forums. I don't use them, but there are plenty of folk who will take an interest. I contribute here so the developers are a bit less distracted from their task.

klinwer, 2024-12-03:
Ok, let me close it.

closed 2024-12-03 by klinwer

klinwer, 2024-12-03:
Hi Dave T.

 

Underextrusion give me the proper hint. 😊

 

I had to tighten the screws on the Idler-swivel assembly with two turns.

 

Now it working as expected.

 

Thanks for your help.

 

Regards

u89djt, 2024-12-03:
Oh, that's interesting. Props for extrapolating successfully! So the oozing was mostly failure to retract? The right thing to do now is to look for mentions of oozing due to retraction not succeeding because of a loose idler, and post something in a good spot if you can't find any.