prusa3d/PrusaSlicer #13700: Anzeigeflattern

Mogli-2000, 2024-12-01:
### Description of the bug

beim Slysen von aufwendigeren Projekten beginnt die Anzeige zu flackerm mit der Invo - keine Rückmeldung -
Das Programm nimmt dan auch keinerlei Befehle mehr an.
Zum teil stundenlang. Oft hilft nur ein zwangsschließen des Programms. 
ich hab auch mit dem Befehl Druckpause ein Problem das mein Drucker diese nicht macht Sovol SV04
mein Rechner ist ein Intel(R) Core(TM) i3-2370M CPU @ 2.40GHz   2.40 GHz mit 16GB Speicher
[image]

### Project file & How to reproduce

https://www.thingiverse.com/thing:6753345

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.9.0 alpa1 win 64

### Operating system

Window 10

### Printer model

https://www.thingiverse.com/thing:6753345

fe60, 2024-12-05:
Seit Version 2.8.0 gibt's Probleme mit der Leistung des PrusaSlicer, die in der 2.9er Version scheinbar noch nicht behoben sind.
Hast du mal die 2.7.4 ausprobiert (siehe https://github.com/prusa3d/PrusaSlicer/issues/12937#issuecomment-2327086681)?

Mogli-2000, 2024-12-05:
Mit 2.7.4 klappt es. Wollte aber auch von den neuerungen profitieren...

fe60, 2024-12-05:
> Wollte aber auch von den neuerungen profitieren...

Nachvollziehbar, aber da heißt es wohl auf Besserung warten.

Da die Performanceprobleme schon in anderen Tickets adressiert sind (das oben verlinkte war eines der ersten), solltest du einem der bestehenden Tickets folgen (ggf. dem initialen Beitrag noch einen "Daumen hoch" hinzufügen) und deines schließen, Viele Tickets machen die Problemlösung nicht schneller.

Das von Dir angesprochene Ignorieren der Druckpause, würde ich in einem neuen Ticket melden (ist generell sinnvoll ein Problem pro Ticket). Darin könntest du auch erwähnen, ob das mit 2.7.4 auch schon passiert.

closed 2026-09-29 by prusaslicer-bot[bot]