prusa3d/PrusaSlicer #13702: Prusa Slicer crashes after selecting Preview

mhoberhausen, 2024-12-02:
### Description of the bug

I have two objects that I'm trying to print (multicolor). The print has a third object that is attempting to wipe to. When doing so at 100% size, the preview functionality works fine. However, when scaling down to 50%, PrusaSlicer crashes each time.

### Project file & How to reproduce

[tmp.zip](https://github.com/user-attachments/files/17980855/tmp.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.8.1+win64-202409181359

### Operating system

Windows 11 64bit

### Printer model

Original Prusa MK4S MMU3 HF0.4 nozzle

u89djt, 2024-12-02:
If you delete either the cranium or the jaw, it succeeds in slicing.
If you merge the cranium and jaw, it succeeds in slicing, so that's you're workaround while the issue is addressed. The painting is preserved - I learned a thing :).

mhoberhausen, 2024-12-02:
Yeah I noticed that doing a single object is not causing issues. Figured since I could reproduce easily, I would post about it for others who are not able to have a temporary workaround

u89djt, 2024-12-02:
Exactly the right thing to do - it's generally worth squeezing as much detail about what you've tried around it to save analysis time.

closed 2026-09-29 by prusaslicer-bot[bot]