prusa3d/PrusaSlicer #13703: Windows 2.9.0-alpha1 Using Arrange with MMU3 profile places purge tower even if not needed.

Bpendragon, 2024-12-02:
### Description of the bug

In 2.9.0-Alpha 1 the purge tower is always treated as an object during arranging if using an MMU3 profile, even if the print is single extruder.  This often leads to somewhat silly arrangements where there is nothing in the middle of the board. 

Actual output 
2.9.0-a 1 screenshot: 
[image]

Expected Behavior:
In 2.8.x it will arrange the objects nicely in the middle of the board. 

2.8.1 latest screenshot:
[image]

### Project file & How to reproduce

[christmas fidgets.zip](https://github.com/user-attachments/files/17983382/christmas.fidgets.zip)

To Reproduce: 
1. Open in 2.9.0 Alpha 1
2. Slice Now (to tell the system it's single color and does not need the purge tower)
3. Back to 3D editor viewer
4. Select either "Arrange" option. 

Actual Ouput: an arrangement with a ghost purge block in the center of the print bed. 

Repeat process in 2.8 and the output is as expected with no ghost purge block.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.9.0-alpha1

### Operating system

Windows 10 22H2

### Printer model

Mk4s w/mmu3 and HF nozzle, using slightly custom profile to speed up initial print start.

clintonefisher, 2024-12-21:
> (From a fellow user following the hub) You can stop it happening by disabling the wipe tower manually in the settings: ![image](https://private-user-images.githubusercontent.com/75209005/391883027-238f8995-fd50-4f9b-a5df-d2367faa0983.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ4MTkyNTEsIm5iZiI6MTczNDgxODk1MSwicGF0aCI6Ii83NTIwOTAwNS8zOTE4ODMwMjctMjM4Zjg5OTUtZmQ1MC00ZjliLWE1ZGYtZDIzNjdmYWEwOTgzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjIxVDIyMDkxMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWYyNzg0YTA1OWZiMWIzZTg0MzI2Y2Y5YTQzYThmY2ZlMmY0YzUxOTdiYTljZWIyMjYwY2NiNjEwNTc5NDQ2ZjUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.yzFCaNUitQKslaJ08Ao5Gf5qJOmlcq8rAg-MxQrfWJI) My impression (for what it's worth) is that this isn't a bug because when the wipe tower functionality is enabled, it's a good idea to get an impression of how much space will be required if and when colour changes are introduced. If you're not planning colour changes, disable the tower.

Thank you @u89djt, I was about to open a bug on this and didn't realize that I could just disable the tower and it would disappear and not interfere with auto-arrange.

Being able to disable the tower this way is good enough for me.  :)

Cheers!

Jan-Soustruznik, 2025-04-09:
Hello, @Bpendragon, thank you for our report. Reported issue is already fixed. Please use the latest PrusaSlicer release 2.9.2. Thank you have a nice day. Issue closed as solved.

closed 2025-04-09 by Jan-Soustruznik