prusa3d/PrusaSlicer #13710: SOLUBLE FULL and SOLUBLE INTERFACE print settings gone?

Sebastian1989101, 2024-12-03:
### Description of the bug

Not sure what changed (used it two weeks or so ago?) but now my PrusaSlicer seems to have removed SOLUBLE FULL and SOLUBLE INTERFACE print settings. I usually used this to use PETG as "Soluble" support for PLA... 

See: https://i.imgur.com/FgYD3Xc.gif

### Project file & How to reproduce

Empty project

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.1+win64

### Operating system

Windows 11

### Printer model

MK4S with MMU3

chocobodingo, 2024-12-03:
I spoke with support about this the other day. I solved it temporarily by rolling back the Buddy3d filament config. Configuration->Configuration Snapshots

rtyr, 2024-12-03:
Those profiles were removed intentionally because it was confusing for some users (since soluble filament profiles are not available yet).

You can make "soluble" profile within any print profile, just by setting contact Z distance to 0 (support settings) and then set appropriate extruder at "multiple extruders" tab.

We will likely return those "soluble" profiles in next update, because it was apparently used by some users.

rtyr, 2024-12-12:
Soluble profiles were added back by configuration update 2.1.4.

closed 2024-12-12 by rtyr