prusa3d/PrusaSlicer #13714: Not deleting supports

AlexElia21, 2024-12-04:
### Description of the bug

I painted with my right click as usual and support is still there. Its the first time I encounter this error. The slicer is still very much usable, its just a bit annoying with this bug haha.

### Project file & How to reproduce

[BugWithSupports.zip](https://github.com/user-attachments/files/18002228/BugWithSupports.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.1 Windows 64

### Operating system

Windows 11

### Printer model

Ender 3 V2 NEO

TerPuc, 2024-12-04:
Dear @AlexElia21, thank you for your report. I sliced the attached 3mf file and there are no supports generated in the area you painted the support blocking. Could you provide us with screenshot? We are not able to reproduce this or maybe we missunderstood your issue. Thank you!
[image]

AlexElia21, 2024-12-05:
Hi Tereza,
No, you are absolutely right. I sliced the file again today and this time
it didnt have any errors. I guess that as it is the first time Ive seen a
bug in PrusaSlicer i didnt even think about restarting the slicer, haha.
Im really sorry, didnt mean to waste your time.
Thanks, Alex

closed 2026-09-29 by prusaslicer-bot[bot]