prusa3d/PrusaSlicer #13715: Black screen of death on the plater page

Parkweld, 2024-12-04:
### Description of the bug

Updated to the latest firmware. 
[image]
Cannot view the stl files on the plater page. Must slice to view any imagery. Cannot make adjustments on layout or modiify the current positions of multiple parts.

### Project file & How to reproduce

[corne case_left_v1.3.001.zip](https://github.com/user-attachments/files/18012524/corne.case_left_v1.3.001.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.1 win64

### Operating system

Windows 11

### Printer model

Prusa MK3S+

TerPuc, 2024-12-05:
Dear @Parkweld , thank you for your report. We were not able to reproduce this state. May I kindly ask you to provide us with your system info? _(Help → System Info)_ Thank you!

Parkweld, 2024-12-05:
[image: image.png]

closed 2026-09-29 by prusaslicer-bot[bot]