prusa3d/PrusaSlicer #13716: Rough top layer on pineapple model

christianlavoie, 2024-12-05:
### Description of the bug

Top layer as a sandpaper-y feel on the leftmost (when looking at printer from the front) part, about 10% of the top layer. 

Original model from https://cults3d.com/en/3d-model/home/sliced-ananas all pieces except the bottom slice have printed fine on the printer. The misprint was neither the first not the last to print, and has been printed incorrectly 3 times now... incl. two instances at the same time, upside down from one another

Filament is Prusament (conveniently enough) _Pineapple Yellow_ :P

Printer is MK4S (factory assembled MK4 upgraded with kit)

3MF file: [B_Slice_01.zip](https://github.com/user-attachments/files/18026146/B_Slice_01.zip)
BGCode file: [B_Slice_01_0.4n_0.1mm_PLA_MK4S_5h21m.zip](https://github.com/user-attachments/files/18026767/B_Slice_01_0.4n_0.1mm_PLA_MK4S_5h21m.zip)

### Project file & How to reproduce

Slice, print, get disappointed :(

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version: 2.8.1 Build: PrusaSlicer-2.8.1+flathub.org

### Operating system

(flatpak on ubuntu) Operating System: Unix System Architecture: 64 bit System Version: Linux 5.19.0-46-generic x86_64

### Printer model

MK4S (factory assembled MK4 upgraded with kit)

christianlavoie, 2024-12-05:
[image]

u89djt, 2024-12-05:
(From a fellow user following the hub)
Referring to your project file, it looks like your photo is of the bottom of one of the parts rather than the top.
[image]
If that's right, then the bed hasn't been successfully probed, or maybe something made it impossible to compensate. 
Pushing out ideas: Is there something stuck under the removable magnetic sheet? Does it print ok if you move it to the back of the bed?

u89djt, 2024-12-05:
It could also be a patch of grease on the build plate. That region of filament not adhering as well might mean it's pushed around by extrusion within it or above it, but the whole thing stays stationary because of the rest of the body being well-adhered.

christianlavoie, 2024-12-05:
It's the toplayer of the bottom slice of the pineapple that's misprinted. 

The project file is the same object (the bottom slice), one "normal" side up, one upside down. In both cases, their toplayer is misprinted.

I checked the bed for bumps and cleaned it with IPA before printing the project. So, possible it's grease or gunk, but doubtful :/

The printer has printed correctly before and after that print (though different models). So far the only thing in common is that specific STL file.

I'll do another print later on, move it a few times and rotate it to see if if anything changes.

I'm not 100% sure where the problem is. The slicing looks good in path view, and as far as I can tell the printer is fine physically speaking (but can I ever be truly sure? :/ ) -- so what's left? G-code generation? Firmware execution?

u89djt, 2024-12-05:
Ohhh right, sorry. I jumped when I saw the same shape within the box you put in the file.
So the key thing is to maintain a consistent picture of where everything is. What does a photo directly top-down of the plate with both objects on it look like? Where it happens in the printer's coordinate system seems likely to be important. Maybe use a filament with better visible contrast. I have a big stack of grey reels for precisely that reason while I'm deriving efficient/stable slice parameters.
We can compare notes on cleaning - if the paper towel or cloth doesn't make a creaking noise as you push it slowly along the plate, the grease hasn't gone.

rtyr, 2024-12-05:
Based on the photo, it looks like typical effect of warped print. Basically part of the object lifted from the bed during the print.

u89djt, 2024-12-05:
Ah, so adhesion and hence cleaning? It's PLA - my less extensive experience hasn't met PLA warping during the print. Afterwards, yes. Maybe I struck lucky with cleaning technique from early on.

u89djt, 2024-12-05:
I'm wrong - I've had PLA warping during prints: organic supports warping is the only explanation I have for a couple of clashing failures where there were no gaps of overlaps visible in the slice preview of the supports. Is there an article somewhere on relative warpyness of infills and supports? Could dropping breaks into infill ease warping for some scenarios? Gyroid is going to tug all the way up the body in every direction, I guess?

rtyr, 2024-12-05:
Even though warping issues are not so common with standard PLA prints, it can still happen. This print uses 5 perimeters, so there is a little bit stronger overall pulling force. As for this particular case, it is better to contact our tech-support via live chat at www.prusa3d.com and discuss possible adhesion issues/sheet type selection, etc.

u89djt, 2024-12-05:
Lots to learn.

themanyone, 2024-12-08:
Topic hasn't yet mentioned an enclosure.

A drafty workplace can make parts shrink, creating wrinkles on the most flexible (top-most) layers.

closed 2026-09-29 by prusaslicer-bot[bot]