prusa3d/PrusaSlicer #13717: Save/Import custom filament profiles to Connect disk space

stur0067, 2024-12-05:
I'd like to be able to see my custom filament profiles across multiple computers without manually moving ini files around.
The disk space provided to Prusa Connect users would be the perfect way to accommodate syncing them across multiple computers.
Even if it was a somewhat manual process to "save to connect" as an icon and the import process required opening up Add/Remove filament configuration wizard. If we could just point to our Connect space like we would do for "Local sources" it would pretty efficient.

mix579, 2024-12-05:
Sounds reasonable. I currently use OneDrive and have moved my config directory over there, which works great but needs some local configuration which may be too much for most "normal" users. This suggestion would be a lot more attractive.

PlumpCat19, 2024-12-05:
This would be pretty useful for someone who shares a printer and uses multitude of different types of filaments.

RWL-69, 2025-01-20:
This is a decent request for many.  It would be a good idea to add the ability to select the config directory for Prusa Slicer in the configuration folder.  There are threads on the Prusa Forum for workarounds that some report work.  It would be easier for users to be able to just select the config directory from a menu.

Sub selections for each of the configuration directories would be even better.

Most modern software that I use has the ability to select your configuration directories to allow shared configuration profiles.

As mix579 says, they are using a cloud drive for configuration space.

closed 2026-01-20 by Jan-Soustruznik