prusa3d/PrusaSlicer #13721: Incorrect Fan Speeds on Multi Material Print from Prusaslicer 2.8.1

Andrew2087, 2024-12-06:
### Description of the bug

As the title states, Prusaslicer is incorrectly generating fan speeds for my multimaterial print up until the first instance of a tool change. I have an IDEX 2 tool (it uses extruders 2 and 3 in prusaslicer), RepRap printer and am trying to print recycled PETG with PLA as the support interface. Despite the settings used for the recycled PETG cooling, it does not turn on correctly until the first instance of a tool change (see the reference 3mfs and photos). If I select both heads to be printing with recycled PETG, then the fan speeds generate correctly. 

### Project file & How to reproduce

The problem occurs in the PETG-PLA project, but not the PETG-PETG one

[Fan Problem Multimaterial.zip](https://github.com/user-attachments/files/18031032/Fan.Problem.Multimaterial.zip)
[image]
[image]
[image]
[image]
[image]
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.1

### Operating system

Windows 10

### Printer model

IC3D Virago 700 (IDEX/Reprap)

hejllukas, 2024-12-17:
Hi, thank you for your report. I was able to reproduce your issue.
I created an internal issue, no. SPE-2628.

lukasmatena, 2025-02-18:
Should be fixed in [PrusaSlicer 2.9.1-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.1-alpha1). Closing.

closed 2025-02-18 by lukasmatena