prusa3d/PrusaSlicer #13723: 3FM project opening

Advicerus, 2024-12-06:
### Description of the bug

when I downloaded project from printables that i wanted to print, but it was in format 3mf. it asked me if I want all project or only 3D model, so I opened only 3D model but some features still remain, so I printed model and it was really really bad, so I changed the nozzle because I was thinking that maybe that was that problem but no. when you open 3mf file and want only model the variable layer height still remains and that was the reason why the print sucks 

### Project file & How to reproduce

donwload 3mf file as such (https://www.printables.com/model/307406-magsafe-iphone-and-airpod-stand/files) open it at prusa slicer and press I want only model [image] and than look at layer height and there it is 
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.1

### Operating system

windows 11 home

### Printer model

Prusa Mini+

neophyl, 2024-12-06:
Well delete the vlh.  Simple.

Advicerus, 2024-12-06:
well if you don't know this and you want to just open 3D model than you should only open 3D model and not model with vlh in it... I printed 5 of these until I realized that the vlh is on...

neophyl, 2024-12-06:
Sorry but its clearly shown in the right hand tree view.  Just like any other modifier would be.  Like a negative mesh used to add a hole to the part.  Its still part of the model data which is what you are loading.  People would complain if it automatically removed those.

Its not loading the profile data so doing exactly what I would expect.  

So not a bug.  
You could convert to a feature request, asking for more control of what to load when importing 'model' data.

u89djt, 2024-12-07:
(Fellow user following the hub)
Disagreeing with @neophyl 
Variable layer heights are chosen to work well with a specific printer, filament, temperatures, cooling and slicing regimen. They are not intrinsic to the shape of the printable object.
Certainly it can be handy to be provided with a good choice of layer heights if they are going to work. For that, you import the project and modify whatever else you need to care about.
The variable layer heights indicator does indeed show that variable layer heights have been set. They shouldn't have been set.
As I understand it, users are only fundamentally expected to divert their attention from exporting gcode to respond to warnings and errors, and mostly to activate supports.
All that said, the devs know more than we do, so I'm looking forward to hearing why the vlh is really imported in this case :)

themanyone, 2024-12-09:
0.5 is a pretty high layer height for 0.4 nozzle. No wonder it wouldn't print.

Prusaslicer uses the printer profile to check whether the current or selected printer is capable of printing the layer heights that come pre-sliced in the model file. But the check is defeated if the max height in the printer profile is set artificially high. 

- You want to set max height lower, so Prusaslicer can warn you when a model is not printable.
- Unfortunately, no such warning is currently implemented.
- So if you put in a lower max height in printer profile, Prusaslicer will just assert->fail and exit. (I am trying to address the assert in PR #13691)
- To work around these problems, users have been setting their printer's max height to a high value like 1mm, just so they can open the .3mf file
    - which causes the mess you experience when trying to print the model sliced with 0.5mm layer heights.

Hopefully, in the future, you can set max height in printer profile to something sane, like 0.35. Then, when it gets implemented, a warning message should then pop up, telling you to re-slice. Again, perhaps, if they accept my pull request #13691. Meanwhile, you can try my code and see if it works. :)

closed 2026-09-29 by prusaslicer-bot[bot]