prusa3d/PrusaSlicer #13728: Auto center configuration with multiple pallets/plates

malamaker, 2024-12-07:
### Description of the bug

When auto center is on, the auto center configuration option will center the pieces to all 9 pallets/plates vs the parts being dropped to the single pallet/plate.

### Project file & How to reproduce

Turn on auto center
Have parts spread out and then try to arrange them to a set of multiple pallets.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0-alpha

### Operating system

MacOS 14.4.1

### Printer model

Creality K1 / Creality K1 Max

arantesxyz, 2025-02-27:
Yeah same here and it kinda sucks. I can't auto layout on a single plate.

lukasmatena, 2025-02-28:
Auto center was removed in [2.9.0-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-beta1). I believe it effectively solves this issue. Closing.

closed 2025-02-28 by lukasmatena