prusa3d/PrusaSlicer #13729: Select different filaments for different beds

jmichelp, 2024-12-07:
**Is your feature request related to a problem? Please describe.**

Prusa Mini+ doesn't support multiple extruder, yet for complex assemblies, I would like to have a single 3MF project with different material for multiple printing beds.

**Describe the solution you'd like**

For complex assemblies, I would like to define the filament being used per printing bed. So that in a single 3MF project, I can group all TPU pieces together in a single printing plate while having PETG or PLA pieces together on other printing plates, etc.
This should already be supported for multi-head printers such as the Prusa XL on a single plate but the mini doesn't support multiple extruders.

**Describe how it would work**

For single extruder printers such as Prusa Mini+, changing the printing filament should only affect the active bed, allowing to group beds per printing material and export it as a single 3MF project.

**Describe alternatives you've considered**

Defining the material per object could be an alternative and on single extruder the slicer would have to warn about mixing objects. It could also allow the Arrange function to automatically create the beds.
But that could be a much more complex change to make or give a more complex UI.

**Additional context**
N/A

themanyone, 2024-12-08:
+1 it would be a good idea to have filaments in printer profiles. Or better yet, a default .3mf file for each printer.

That way, when you change printers, it loads filaments, print settings, layer height, etc. from that printer's default .3mf file.

lukasmatena, 2026-09-01:
Implemented in [3.0.0-alpha11](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_3.0.0-alpha11). Closing.

closed 2026-09-01 by lukasmatena