prusa3d/PrusaSlicer #13730: [PrusaSlicer 2.9.0-alpha1] Fuzzy Skin on top layer (smart fill)

juckendesAuge, 2024-12-07:
### Description of the bug

Top layer (which was marked with fuzzy skin) is not printed. In the gcode preview it is also displayed open, so that you can see the infill.

[image]

[Fuzzy Skin Top Layer without Layer Bowl-Without-Centering.3mf.zip](https://github.com/user-attachments/files/18049413/Fuzzy.Skin.Top.Layer.without.Layer.Bowl-Without-Centering.3mf.zip)

### Project file & How to reproduce

Open 3mf file. Slice it. Watch sliced model.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.9.0-alpha1 (flatpak)

### Operating system

Fedora release 41 (Forty One)

### Printer model

Prusa MK3S+

juckendesAuge, 2024-12-07:
Please Close it. Setting Top Solid layers was on 0 ;)

themanyone, 2024-12-08:
There should be an option to close this yourself. It gives me an idea for "fuzzy skin on top layer". I want to try randomly bumping z up and down on top layer to see if it is possible.

juckendesAuge, 2024-12-08:
That already exists.

UPDATE! - Non-Planar Top Fuzzyskin - 3D Printing Guide! - https://youtube.com/watch?v=85FJl5P0AoU

GitHub - TengerTechnologies/Fuzzyficator: (Work In Progress) A Gcode postprocessing script to add non-planar "Fuzzyskin" to top flat surfaces. - https://github.com/TengerTechnologies/Fuzzyficator