prusa3d/PrusaSlicer #13734: Tilt Build Plate View At Mousepointer, Not Around Build Plate Center

KlausKraemer, 2024-12-09:
**Is your feature request related to a problem? Please describe.**

Tilting the build plate at a bigger zoom factor can be very tiresome, when your model is not centered on the build plate. As tilting the model / build plate in slicer moves around the build plate center, you loose sight of your models easily, when they move out of view in larger zoom factors.

**Describe the solution you'd like**

Tilting the build plate should instead turn around the actual cursor position - even projected on the x-y position of the build plate would improve usability a lot.

Please consider that if one's zoomed in, it's very probable they want to see exactly, what's shown in the zoomed in picture and tilting should alter the angle of perspective on that chosen viewpoint - not move it away!

u89djt, 2024-12-09:
(Fellow user following the hub )
The mouse cursor position is two dimensional, so can't define a rotation point for the view by itself.
I would like to be able to click somewhere on an object and drag to make the view rotate around that three-dimensional point. Maybe we could even set a 3D cursor to be a persistent rotation centre.
Currently, the view rotates around a point on the bed or its extension in its plane where the central axis of the view intersects.
Middle click to translate the view in the plane of the screen.
Ctrl-click lets you click and drag on an object to rotate the view, but it does not pick up any information from the object.

KlausKraemer, 2024-12-10:
> (Fellow user following the hub ) The mouse cursor position is two dimensional, so can't define a rotation point for the view by itself. I would like to be able to click somewhere on an object and drag to make the view rotate around that three-dimensional point. ...

I am aware, that a mouse pointer position is two dimensional. But it would simply help in a first step to use a projection of its positon onto the build plate and use this new point as rotation center. Your suggestion of setting a 3D cursor as rotation center would be superb and also the combinatin with ALT or CTRL key.

Tupson444, 2026-09-19:
Yes, please make the center of rotation around mouse. It is very annoying when trying to rotate view while zoomed in at a small object near the edge of buildplate, or when looking at top layers of a tall print.