prusa3d/PrusaSlicer #13740: supp organique toujours actif

18pool49, 2024-12-10:
### Description of the bug

bien que AUCUN support est demandé , il est IMPOSSIBLE d'utiliser la Ht. variable car une alarme nous indique que les support organique sont actif (( ce qui est faut )) 

### Project file & How to reproduce

cela ce produit avec tout les STL 

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.1 + 2.9.0 a1

### Operating system

W11

### Printer model

XL5

u89djt, 2024-12-10:
"although NO support is requested, it is IMPOSSIBLE to use the Ht. variable because an alarm indicates that the organic supports are active ((which is wrong))"

(From a sympathetic fellow user) Post a 3mf file (zipped) just before you attempt to enable variable layer heights and get the error message. With that, it should be possible begin to troubleshoot your problem.

18pool49, 2024-12-10:
2.8.0 la hauteur variable ce fait correctement 
[image]
2.8.1 la hauteur variable ce fait correctement 
[image]
2.9.0. erreur soit disant support 
[image]
[397staravia 2.9.0 a1 avant HVariable projet.zip](https://github.com/user-attachments/files/18084570/397staravia.2.9.0.a1.avant.HVariable.projet.zip)
voilà le zip avant de demander la Hauteur variable

u89djt, 2024-12-10:
The second screenshot says it's 2.8.0, not 2.8.1?
I get the warning in 2.8.1.
Organic supports are selected in your project file:
[image]
It's a warning that will allow you to proceed, not an error that stops slicing.
The warning reminds you that enabling supports will cause a problem.
If you do enable supports, the slicer shows an error: Erroneous Data

u89djt, 2024-12-10:
I apologise - I get the same result with 2.9. My opinion is not relevant.
[image]

18pool49, 2024-12-10:
comme vous me l'écrivez ,
j'ai activé les supports  qui était en "organique" j'ai modifié en grille  et supprimé l'appel des supports en haut 
je suis revenu en "plateau " et lancer le tranchage et tout c'est bien déroulé pas erreur 

j'ai refais l'opération inverse en support j'ai sélectionné  les supports et choisi "organique" dans la liste  
et ensuite 
j'ai désactivé les supports en haut >>> mais le choix "organique" bien que masqué  (ombré) reste toujours actif et bloque la hauteur variable et donne cette alarme  >>
 j'ai donc effacer la sélection "organique" changé par "grille" et désactivé les supports en haut  >> "grille " est ombré 

et la découpe du slicer se fait bien en hauteur variable 

la sélection "organique" même "OMBRE" reste active quand on choisi la hauteur variable  
!!!  c'est loin d'être logique  !!!!

u89djt, 2024-12-11:
I agree that the logic in this specific scenario doesn't suit the user when it potentially could.
We're seeing that the support type selection being organic causes the slicer to throw an error when an object on the plate has variable layer heights defined for it, whether or not support is automatically generated. We haven't explored what happens if supports are painted on. 
So we imagine that it's either an error, or there are other scenarios that make it necessary for the slicer to do it that we are not aware of. It may be that it's safer this way for future introduction of new methods. It certainly isn't necessary to have the organic type selected while using variable layer heights, so that could be a motivation for pre-emptively disallowing it.
Organic supports aren't the default in any of the print profiles provided by Prusa, so they are effectively treating them as something you should have to actively choose, and therefore take some responsibility for. If it is an error, that could potentially explain why it hasn't been noticed, or why it isn't a priority to introduce a change with concomitant risk of further errors being introduced. Developing software to put in the hands of users is bewilderingly fraught with surprises.
I think I don't know enough for further speculation to be of value, so since you can proceed with slicing and printing, I'll bow out.

18pool49, 2024-12-11:
bonjour 
ma demande est une remarque d'un simple utilisateur privé  ; qui est pas programmeur en informatique 
comme l'on dit communément "" un bricoleur "" du dimanche 
mais cette difficulté , cette alarme est pas pour simplifier la compréhension du slicer pour le commun des mortels ; dont je fais parti 
moi personnellement je vais appliquer la règle que nous avons découvert ou esquiver cette alarme qui est pas une vrais alarme mais une anomalie de programmation 
pour moi le problème est résolu mais pour les autres utilisateur de ce Slicer il subsistera une certaine amertume et incompréhension
merci pour votre implication et votre patience 
salutations

closed 2026-09-29 by prusaslicer-bot[bot]