prusa3d/PrusaSlicer #13743: Sort settings in the list displayed for the "Add settings" contextual command

EricPobot, 2024-12-12:
It would be nice if the settings list was sorted alphabetically.

Currently, the sequence is displayed without any obvious order, which makes it tedious to find the desired one(s) when there are many. For instance, here is what appears when selecting "Layers and Perimeters" in the "Add settings" menu :

[image]

fe60, 2024-12-13:
Sorting the entries alphabetically adds at least some structure. But I think this would still not be a good solution. "Bottom solid layers" would be far away from "Top solid layers".

I propose to arrange the entries in groups similar to the regular settings and to order them similar to the regular settings.

EricPobot, 2024-12-13:
@fe60 I agree with your point obviously, but alphabetic ordering has several selling points : 
- it is enough for quickly finding the searched item in a long list
- it is neutral, since grouping risks to be based on criteria that could be non obvious in some cases (I've no example on top of my mind, but I'm pretty sure there are some)
- above all, it is easy to implement

It could thus be quite significantly better than the current situation in which the items seem to be listed in a random sequence and I'll be more than happy with this option.

closed 2026-09-29 by prusaslicer-bot[bot]