prusa3d/PrusaSlicer #13746: [Multi-material] Slow down print when printing on top of different material

andyylin, 2024-12-13:
**Is your feature request related to a problem? Please describe.**
My wipe towers and supports are a mess when I print PETG on PLA if I don't slow them down.

[image]

**Describe the solution you'd like**
Option for print to slow down when putting down a layer on top of a different material. So for example, slow down PETG when printing on top of PLA but keep speed normal when PETG is printed on PETG. 

**Describe how it would work**
Detect that layer below print head move is a different material and set to a slower speed (adjustable in settings). 

**Describe alternatives you've considered**
Slowing down the entire print is inefficient. 

**Additional context**
Wipe tower also needs this as the wipe tower often fails and I'm left with a spaghetti mess.

bobbyh41, 2024-12-14:
I had this same problem. It’s been a while, but I think I did a height range modifier for just the interface layer and slowed the printer way down. Then, I returned it to normal after the interface. It was a simple print that allowed me to just do it in one place, but I agree that a setting that slows down the first layer on top of a dissimilar material setting would be great.

jhnphm, 2025-03-26:
I'm also having this problem w/ soluble supports, where the actual material being supported peels off due to insufficient adhesion.

u89djt, 2025-03-26:
@jhnphm are you using a soluble filament? Is it a soluble filament that should stick to the main print? Nobody knows exactly what you're dealing with till you post a 3mf file with your problem. You need to zip 3mf files to post them.

jhnphm, 2025-03-27:
[wings.zip](https://github.com/user-attachments/files/19480659/wings.zip)
Basically PLA to BVOH adhesion was somewhat poor and didn't stick as much as it should, and the wing would peel off the support. Slowing down helped but takes forever.

Note: 
My actual problem in this case was the tool offset being off causing underextrusion/insufficient squish on top of the support, and recalibrating that after cleaning helped significantly. Slowing down the entire solid layers did help somewhat but didn't solve the final problem

closed 2026-09-29 by prusaslicer-bot[bot]