prusa3d/PrusaSlicer #13748: PrusaSlicer crashes on MacOS when attempting to sign in with Prusa account.

ElenaSore, 2024-12-13:
### Description of the bug

Upon opening PrusaSlicer and attempting to log in, the app completely locks up until I force quit. Currently on the 2.9.0-alpha1, however this issue also existed in the same form within 2.8.1, and I updated in hopes of the new version fixing it.

### Project file & How to reproduce

[Untitled.3mf.zip](https://github.com/user-attachments/files/18129183/Untitled.3mf.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.9.0-alpha1-macOS

### Operating system

macOS Sequoia 15.2

### Printer model

Prusa XL 5T, MK3S, MK3S+

poolrev, 2024-12-14:
The same issue occurs on my Windows 11 notebook. I've tried versions 2.8.1 and 2.9.0 beta1. Unfortunately both versions stop responding once I click the "Log in" button.

user0815852, 2024-12-14:
Same here... please fix it

SalDaj2022, 2024-12-14:
I am Facing the same issue with both 2.8.1 and 2.9.0...

nikpawlak, 2024-12-14:
Add me to the list. 
I have tried uninstalling. Deleting cookies. Reinstall.
Deleting Prefrences. 

Now the app hangs upon opening. Can't make it through the congfig. wizard unless I disable the network. 

PrusaSlicer 2.8.1 and 2.9.0 beta.  
Mac OS 15.2
M1 Max MBP

arekm, 2024-12-15:
Here freshly dowloaded version (just now) locks ups (the macos rainbow spinning wheel appears) when I click on "Log in" menu. macos sequoia 15.2.

But the same thing happens with 2.8.1 which was working a few weeks ago.

Some sequoia 15.2 change or prusa server side change?

sdelmas, 2024-12-15:
Same here, I am on 2.9.0-beta-1 but only went to that from the prior version to see if this fixes the spinning ball of doom. I did a complete uninstall (including removing Library and cache files manually). The printer is available and can be accessed through the portal. 

My mac is an intel macbook pro (plenty of diskspace and ram and i9 processor. It's on the very latest Mac OS with all patches applied (15.2).

I noticed that I have to wait about 2 minutes and then the app will be fine. This feels very much like a timing out network request (just guessing, what do I know ;-). Let me know if there is something I can do to help.

legendontour, 2024-12-16:
Also happening to me.  Has happened on 2.9 - alph and beta.  Tried re installing 2.8.1 stable release and still happening.  Even if I don't sign in, happens when I try and use configurations assistant to add the new high flow nozzle.

kubispe1, 2024-12-16:
Hi,
On Thursday, we experienced unexpected downtime with PrusaAccount (https://github.com/prusa3d/PrusaSlicer/issues/13745). Although the system has since been relaunched, we are aware that some users are still encountering issues.

Our team is actively investigating the situation to determine the root cause and how to prevent such issues in the future. The department responsible for PrusaAccount is currently implementing changes to improve the situation.

Currently, some users have confirmed that they can sign in again. Please try as well and let us know.(https://github.com/prusa3d/PrusaSlicer/issues/13762).

Sorry for the inconvenience.

arekm, 2024-12-16:
Works for me now but anyway some bug is there in PrusaSlicer as app shouldn't hang forever in case of cloud failure.

user0815852, 2024-12-17:
tested, 2.8.1+MacOS-arm64 login works