prusa3d/PrusaSlicer #13750: PrusaSlicer crashes on MacOS when attempting to upload gcode to printer (2.9.0-beta1)

pipcorona, 2024-12-13:
### Description of the bug

When trying to upload gcode to a physical printer, the application crashes. I can see a message that it is resolving the IP address, and once it resolves it crashes. I was able to export the gcode to my computer and then upload through the browser. I am attempting to upload to a Prusa XL.

### Project file & How to reproduce

Slice a model, attempt to upload to physical printer. This is happening with multiple models.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0-beta1

### Operating system

15.2 (24C101)

### Printer model

Prusa XL

audiorazor, 2024-12-14:
Same issue here.

pipcorona, 2024-12-14:
I forgot to put I have confirmed that PrusaSlicer does have full local network access in the Privacy and Security Settings.

JohnOCFII, 2024-12-14:
Similar issue while attempting to send to a MK4S via PrusaConnect.  Saving to GCODE locally on my Mac worked fine.

Steps to reproduce:
- Slice file
- Press `Send to Connect`
- Select Printer -- Confirm
- Boom!

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               PrusaSlicer [5008]
Path:                  /Applications/Other Apps/PrusaSlicer-2.9.0-b1.app/Contents/MacOS/PrusaSlicer
Identifier:            com.prusa3d.slic3r/
Version:               PrusaSlicer PrusaSlicer-2.9.0-beta1 (PrusaSlicer-2.9.0-beta1)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-12-13 18:20:09.8942 -0600
OS Version:            macOS 15.2 (24C101)
Report Version:        12
Anonymous UUID:        E1A9B3E8-17C0-3635-633F-31D076EC9805

Sleep/Wake UUID:       00285188-CC3A-4EB4-85FE-F7BA32D43519

Time Awake Since Boot: 22000 seconds
Time Since Wake:       1383 seconds

System Integrity Protection: enabled

Crashed Thread:        0  slic3r_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 6 Abort trap: 6
Terminating Process:   PrusaSlicer [5008]

Application Specific Information:
abort() called

Thread 0 Crashed:: slic3r_main Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x18b39b720 __pthread_kill + 8
1   libsystem_pthread.dylib       	       0x18b3d3f70 pthread_kill + 288
2   libsystem_c.dylib             	       0x18b2e0908 abort + 128
3   libclang_rt.asan_osx_dynamic.dylib	       0x10e1ce1bc __sanitizer::Abort() + 68
4   libclang_rt.asan_osx_dynamic.dylib	       0x10e1cd884 __sanitizer::Die() + 212
5   libclang_rt.asan_osx_dynamic.dylib	       0x10e1b31c0 __asan::ScopedInErrorReport::~ScopedInErrorReport() + 1124
6   libclang_rt.asan_osx_dynamic.dylib	       0x10e1b2498 __asan::ReportGenericError(unsigned long, unsigned long, unsigned long, unsigned long, bool, unsigned long, unsigned int, bool) + 1436
7   libclang_rt.asan_osx_dynamic.dylib	       0x10e19f05c wrap_wcslen + 392
8   PrusaSlicer                   	       0x1082e6324 Slic3r::GUI::PrintHostQueueDialog::on_progress(Slic3r::GUI::PrintHostQueueDialog::Event&) + 2300
9   PrusaSlicer                   	       0x10a8f3054 wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 360
10  PrusaSlicer                   	       0x10a8f2cc8 wxEvtHandler::ProcessEventLocally(wxEvent&) + 88
11  PrusaSlicer                   	       0x10a8f2a90 wxEvtHandler::ProcessEvent(wxEvent&) + 56
12  PrusaSlicer                   	       0x10a8f26d8 wxEvtHandler::ProcessPendingEvents() + 320
13  PrusaSlicer                   	       0x10a8322f4 wxAppConsoleBase::ProcessPendingEvents() + 252
14  PrusaSlicer                   	       0x10a8fc3e4 wxCFEventLoop::OSXCommonModeObserverCallBack(__CFRunLoopObserver*, int, void*) + 84
15  CoreFoundation                	       0x18b4bbbd8 __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 36
16  CoreFoundation                	       0x18b4bbac4 __CFRunLoopDoObservers + 552
17  CoreFoundation                	       0x18b4bb0f4 __CFRunLoopRun + 788
18  CoreFoundation                	       0x18b4ba724 CFRunLoopRunSpecific + 588
19  HIToolbox                     	       0x196a12530 RunCurrentEventLoopInMode + 292
20  HIToolbox                     	       0x196a1817c ReceiveNextEventCommon + 216
21  HIToolbox                     	       0x196a18508 _BlockUntilNextEventMatchingListInModeWithFilter + 76
22  AppKit                        	       0x18f025034 _DPSNextEvent + 660
23  AppKit                        	       0x18f9892d4 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 688
24  AppKit                        	       0x18f018060 -[NSApplication run] + 480
25  PrusaSlicer                   	       0x10a7f9068 wxGUIEventLoop::OSXDoRun() + 140
26  PrusaSlicer                   	       0x10a8fcdb8 wxCFEventLoop::DoRun() + 40
27  PrusaSlicer                   	       0x10a861658 wxEventLoopBase::Run() + 192
28  PrusaSlicer                   	       0x10a8316d8 wxAppConsoleBase::MainLoop() + 208
29  PrusaSlicer                   	       0x10a7b8e2c wxApp::OnRun() + 36
30  PrusaSlicer                   	       0x10a897e94 wxEntry(int&, wchar_t**) + 88
31  PrusaSlicer                   	       0x1078cd688 Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) + 700
32  PrusaSlicer                   	       0x104e6f384 Slic3r::CLI::run(int, char**) + 53740
33  PrusaSlicer                   	       0x104e860f0 main + 656
34  dyld                          	       0x18b054274 start + 2840
```

pgkovakas, 2024-12-14:
https://github.com/user-attachments/assets/da0d6e9c-bfb3-4787-a017-7b7b645107d2

Same crash on mac, MK4S HF 0.4 nozzle

malcinator, 2024-12-14:
Same issue here only uploading to Repetier-Server, worked as expected in the 2.9.0-alpha version.

PGlyre, 2024-12-14:
Same problem for me and I have Apple M1 with macOS 15.2
This problem was not present in the alpha version

SachCZ, 2024-12-14:
Hello, thank you for the report! There has been an issue during the software release mac. **We have accidentally released a testing version.** We have fixed the situation just now. Please re-download the version from https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-beta1

To make absolutely sure it is the correct one:
- **the dmg size is 119.1 MB on disk and 114 MB on github**
- **the sha256 of the correct dmg file is `aef88582fd5414cc34650bd2efcc131346f39d7556cd9988d5b2b3b057d75eb4`** (for advanced users

The incorrect version had around one hundred and seventy four MB (I am intentionally not putting the number here for any just skimming through).
We are very sorry for the trouble caused.

pipcorona, 2024-12-14:
I can confirm the new release file solves the problem.

closed 2024-12-14 by pipcorona

Grifnz, 2024-12-16:
The Beta1 of 2.9.0 is still crashing for me when it tries to login to my connect account.