prusa3d/PrusaSlicer #13752: Allow PrusaSlicer to Download Files from Any Trusted URL by Removing URL Restrictions for Direct Downloads

Maxime-JULLIEN, 2024-12-13:
### Is your feature request related to a problem? Please describe.

I’m frustrated because PrusaSlicer blocks the download of files from external sources by enforcing a restriction to only allow downloads from `https://printables.com`. This limitation hinders the use of PrusaSlicer to open files directly from other trusted sources via direct links, such as:

```
prusaslicer://open?file=https%3A%2F%2Ffiles.printables.com%2Fmedia%2Fprints%2F1075335%2Fstls%2F8244782_c150dc4e-6de0-45db-9efe-0044d34b9180_26c73b9f-e63f-4e93-af8a-f5608af95e13%2Fcable-clip-v1.stl
```

This restriction negatively impacts workflows where files are hosted on private or external repositories, preventing users from leveraging the full potential of direct file integrations.

---

### Describe the solution you'd like

I would like PrusaSlicer to remove the code that enforces this restriction:

```cpp
if (!boost::starts_with(url, "https://") || !FileGet::is_subdomain(url, "printables.com")) {
    std::string msg = format(_L("Download won't start. Download URL doesn't point to https://printables.com : %1%"), url);
    BOOST_LOG_TRIVIAL(error) << msg;
    NotificationManager* ntf_mngr = wxGetApp().notification_manager();
    ntf_mngr->push_notification(NotificationType::CustomNotification, NotificationManager::NotificationLevel::RegularNotificationLevel, msg);
    return;
}
```

Removing this restriction would allow users to download and open files from any trusted URL using the custom protocol `prusaslicer://`.

---

### Describe how it would work

1. **Workflow**:  
   - When a `prusaslicer://open?file=<url>` link is triggered, PrusaSlicer should directly attempt to download and open the file from the provided URL, without checking if it belongs to `printables.com`.  

2. **Error Handling**:  
   - If the URL is invalid or unreachable, display an error notification, but avoid blocking other valid external URLs.  

This change would not compromise security as users are already responsible for verifying the source of their files.

---

### Describe alternatives you’ve considered

- Allow users to whitelist specific domains in the application settings to extend the allowed download sources beyond `https://printables.com`.  
- Provide a configuration option to disable the restriction, either via a flag in the settings or a command-line parameter for advanced users.

---

### Additional context

This feature request aims to improve PrusaSlicer’s flexibility by supporting advanced workflows where file hosting services or custom repositories are used. It would make the slicer more user-friendly and adaptable to diverse use cases.

Jan-Soustruznik, 2024-12-19:
Hello, @Maxime-JULLIEN , thank you for you feedback. We currently have no plans to change this "restriction".

closed 2024-12-19 by Jan-Soustruznik

koenpunt, 2025-04-30:
Now that [thingiverse.com has been added as a valid origin](https://github.com/prusa3d/PrusaSlicer/blob/e70983ad9170c583e0b5d55250740ada1bdd19f7/src/slic3r/GUI/Downloader.cpp#L171), can cults3d.com and other not be added as well?

HairyStuntWaffle, 2025-05-24:
it would be amazing to get this as stndard.

Spice-King, 2025-06-27:
Hey, I'd like this too, as I'm going to be hosting a personal instance of Manyfold as my place to hold my collected mix of STLs (some that are paid, some that are not, and a good mix of stuff that's not even publicly available). I don't _want_ this to be something to consider switching slicers over, because I otherwise like what PrusaSlicer does.

Cassius-create, 2025-06-30:
If 3 nokets a cover anti=hardaway;

bklevence, 2025-08-31:
I too have a manyfold instance and would like to see this, bumping so you see the interest.

Tedderouni, 2025-12-29:
+1 for Manyfold integration

pstaubs, 2026-10-01:
I chose Prusa for my printing ecosystem because it seemingly embraced and supported power users to customise the product to fit their needs. Decisions such as these which build walled gardens fly in the face of that ethos and make me question whether Prusa is still committed to those early founding principles👎.

dlareau, 2026-10-06:
All I wanted is to be able to open a file from my self hosted model library, it is wild that it is hard coded in that you can't do that. Just make allowed domains a configuration setting or something.