prusa3d/PrusaSlicer #13753: Wiping while retract wrong path on single line

eurohell, 2024-12-14:
### Description of the bug

If the object requires it to have one single extraction line, the setting "Wipe while retracting" [wipe[0]) prevents the path of the extrusion to be followed back, but instead it will wipe into any random direction, which leads to ugly strings (left object in example). On extrusions going a full circle (right object in the example or outer lines), everything works fine.
[image]

### Project file & How to reproduce

[Wiping wrong direction on single extrusion line.zip](https://github.com/user-attachments/files/18133845/Wiping.wrong.direction.on.single.extrusion.line.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.9.0-beta1, 2.8.1

### Operating system

Windows 11

### Printer model

MK4

hejllukas, 2024-12-17:
Hi, thank you for your report. I was able to reproduce your issue.
I created an internal issue, no. SPE-2627.

lukasmatena, 2025-02-18:
Should be fixed in [PrusaSlicer 2.9.1-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.1-alpha1). Closing.