prusa3d/PrusaSlicer #13755: Arc Fitting Creates VFAs

spooknik, 2024-12-14:
### Description of the bug

When slicing cylindrical objects of certain sizes, arc fitting does not produce a smooth speed. (see picture). This in turn produces VFAs as the printer is being told to slow down rather than maintain a constant speed throughout the arc. 

Jerk also affects this, if Jerk is raised to 15 instead of the default 8, the problem goes away.  However a jerk of 15 is not ideal for all prints. 

All print profiles except those labeled 'STRUCTURAL' produce this problem. 

[image]

0.25 mm SPEED -  Prusament PLA - MK4S HF0.4 
Top: Arc Fitting On
Bottom: Arc Fitting Off
[image]
[image]

### Project file & How to reproduce

1. New Project
2. Choose 0.25 mm SPEED
3. Add cylinder
4. Resize to 150% 
5. Slice
6. View Actual Speed 

Project:
[Arc_Fitting_VFA.zip](https://github.com/user-attachments/files/18135078/Arc_Fitting_VFA.zip)

Screenshot:
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.1+win64

### Operating system

Windows 11

### Printer model

Prusa MK4S

u89djt, 2024-12-14:
(Fellow user taking an interest)
[image]
The gcode generation resolution is beating against the arc fitting algorithm's scale and the overall radius, and yields arcs that don't line up well enough for the control to create good continuity. If you increase G-code resolution - that is, reduce the magnitude of the acceptable error descriptor - to 0.002mm the speed variation goes away. Why doesn't it happen when arc fitting is disabled? The arc is an approximation to the shape actually present in the data, so there are discontinuities, maybe not in absolute value, but in first and second derivatives certainly, and an arc has to be controlled to a constant second derivative so it takes time/distance to settle down to the new combination of dynamics.
[image]
[image]

u89djt, 2024-12-14:
[Uploading higher resolution Arc_Fitting_VFA.zip…]()
The preview is smooth, but I don't know if the seam will bounce around still. It's still a fast print, so you'd expect echoes. I don't know that the slicer preview makes any attempt to estimate echoes because they can be dependent on the condition of the machine, I think?
If it's still bad, reduce the external perimeters speed?

u89djt, 2024-12-15:
I was spouting plausible garbage. Cranking up the slice resolution is actually stopping the curve fitting happening at all. It's G1 commands all the way round. I didn't pause to move the horizontal slider as it was too "obvious".

u89djt, 2024-12-15:
If I shift over to an XL profile it goes away and I haven't persuaded it to come back yet.
Could it be to do with the relationship between the bed slinging axis and the tool transport axis?
You're getting the requested external perimeter speed while both X and Y are driving, but it's backing off when only X or Y are available. That seems to be motivating division of the arc into small segments, and begins to look like repeatedly setting itself up for a different control regime.
[image]
With jerk cranked up, it doesn't seem to be aiming for your requested external perimeter speed, but layer times are low, so maybe that's why. It's succeeding in carrying momentum all the way round, so time isn't lost to all the green spots of the original.
Yeah, set the minimum un-slowed layer time to zero, and it's obviously aiming for the requested speed, and even with the higher jerk, we're seeing slowdowns:
[jerk15 noslow Arc_Fitting_VFA.zip](https://github.com/user-attachments/files/18139576/jerk15.noslow.Arc_Fitting_VFA.zip)
[image]

u89djt, 2024-12-15:
Same phenomenon on XL input shaper 
[image]

u89djt, 2024-12-15:
It looks it may not be arc fitting per se; rather replacing a sequence of line segments for an extrusion with a different sequence that might be more evenly spaced when the overall progress matches an arc at its vertices.
https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/master/lib/Marlin/Marlin/src/gcode/motion/G2_G3.cpp
* The arc is traced with many small linear segments according to the configuration.

Have I read that right?
So there are no smooth arcs. It seems like you just need to increase jerk or decrease the speed of the external perimeter if you really want to use arc fitting for a potentially smaller gcode file. It seems to be rendering a coarser path than the source model.

u89djt, 2024-12-15:
https://marlinfw.org/docs/gcode/G002-G003.html
Arc moves actually generate several short straight-line moves, the length of which are determined by the configuration option MM_PER_ARC_SEGMENT (default 1mm)

u89djt, 2024-12-15:
https://github.com/MarlinFirmware/Marlin/issues/2519
By default it is 1mm, but you can change this value in Configuration_adv.h to 0.1 if you prefer:
#define MM_PER_ARC_SEGMENT 1

spooknik, 2024-12-15:
> https://marlinfw.org/docs/gcode/G002-G003.html Arc moves actually generate several short straight-line moves, the length of which are determined by the configuration option MM_PER_ARC_SEGMENT (default 1mm)

It might be this variable that needs to be lower then. 

> The arc is traced with many small linear segments according to the configuration.

If you look at a G3 arc it's one command but is broken down into small segements, so without diving into the code too much I would guess that's how it knows how to make the arc. (It reminds me of how Archimedes measured PI the first time.)

u89djt, 2024-12-15:
I'll be making sure arc fitting is deactivated in all my profiles to avoid pointlessly grotty surfaces.
Are there any circumstances that would motivate activating it?

spooknik, 2024-12-16:
If you want a smaller gcode file then arc fitting should stay on. If you're using Structural then it also seems like arc fitting doesn't produce the 'small segments' speed difference.

u89djt, 2024-12-16:
Right, and if your source mesh is finer than 1mm, you'll lose smoothness in the surface. A tradeoff if storage space or data transfer rate is an issue?
Have you printed your object in a structural profile to compare with the original? Less ringing, same flat spots, I assume.
Obviously we're wishing the segment length could be a setting we have access to. It's been queried since 2015 at least, and it's an obvious preference, so I presume it's been found to create difficulties.

u89djt, 2024-12-16:
Here's discussion of arc fitting in this repository: https://github.com/prusa3d/PrusaSlicer/issues/24 
Someone made a judgement call in enabling it by default in some profiles, like the XL. 
Maybe it tames the slicing and export task on less powerful computers? There must be pathological cases we're not going to spot by guessing.

spooknik, 2024-12-16:
> Have you printed your object in a structural profile to compare with the original? Less ringing, same flat spots, I assume. Obviously we're wishing the segment length could be a setting we have [...]

Actually, it's a bit counterintuitive, but usally the faster you print the less ringing and increase in surface quality you get. All printers have some resonance and some speeds produce a lot and others produce very little. You can see this if you [run a VFA test](https://www.printables.com/model/1053867-vfa-test-gcode-for-mk4s), for me and most MK4/s printers i've seen the faster the better.  I would expect worse surface quality on the structural profile because it runs at a speed with more vibrations.

u89djt, 2024-12-16:
Huh. Maybe the tool's momentum or play in the structure becomes significant in comparison to the authority of the servos at higher speeds, and so the tool will tend to carry on regardless through kinks in shallow turns or straight lines? The lighter the tool the higher the speeds at which VFAs persist? I've completely lost sight of my wheelhouse :)

JetroDR, 2025-11-08:
Looks like the latest implementation of "Junction Deviation" in 2.9.4 is also a solution to this problem.
Slicing it on the Core One L profile does the trick. Hoping this parameter will be implemented soon in other profiles soon aswell!

closed 2026-09-29 by prusaslicer-bot[bot]