prusa3d/PrusaSlicer #13757: Login not working on Mac

Kai360-dot, 2024-12-14:
### Description of the bug

When trying to login into the prusa account from Mac via prusa slicer the login window doesn't open.
Login from within is currently not possible.

### Project file & How to reproduce

[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.9.0-beta1

### Operating system

MacOS 15.1.1

### Printer model

Prusa MK4

TravisWilder, 2024-12-14:
the login works, but everytime PS crashes after uploading a job to the printer

JohnOCFII, 2024-12-14:
If you downloaded the Beta more than 10 hours ago, it is likely this issue: https://github.com/prusa3d/PrusaSlicer/issues/13750#issuecomment-2543089390

Prusa Research has re-uploaded a fixed Mac Beta version.  Grab a fresh download and try again.

olsonjj, 2024-12-15:
Same issue here in both the released 2.8.1 and in latest 2.9.0 beta.   I try to log into my Prusa Connect from Prusa Slicer and nothing happens, then beach ball of death. My only option is to force quit the app.  I've restarted my computer to verify and still doesn't work.  

I'm also running Mac OS 15.1.1 so maybe there is commonality here.

olsonjj, 2024-12-15:
> If you downloaded the Beta more than 10 hours ago, it is likely this issue: [#13750 (comment)](https://github.com/prusa3d/PrusaSlicer/issues/13750#issuecomment-2543089390)
> 
> Prusa Research has re-uploaded a fixed Mac Beta version. Grab a fresh download and try again.

The linked issue is crashing on upload. The issue that @Kai360-dot  is seeing and myself is inability to log in.

olsonjj, 2024-12-15:
@Kai360-dot  - mine is just working again.  I'm guessing there was a fix on the PrusaConnect side.

Kai360-dot, 2024-12-16:
Yeah it got fixed! Thanks a lot!

kubispe1, 2024-12-16:
Thanks for confirmation. here is my comment to the situation:  https://github.com/prusa3d/PrusaSlicer/issues/13758#issuecomment-2547065477
Sorry for the inconvenience.