prusa3d/PrusaSlicer #13758: Prusaslicer 2.8.1 login not working on windows 11

bradylobue, 2024-12-14:
### Description of the bug

When trying to login on Prusaslicer 2.8.1, it does not work. Once the login button is clicked the mouse just spins until you have to close the program due to program not responding

### Project file & How to reproduce

Open Prusaslicer
Click on login 
program does not respond

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.1+win64

### Operating system

Windows 11 Home 3/17/23 OS Build 22631.4602

### Printer model

PRUSA MK4S

MindlessDupe, 2024-12-14:
I have the exact same problem - no popup for credentials, nothing. I first noticed this a couple days ago. The same thing also happens when I run the Configuration Wizard, except there I get the opening window from which it hangs in the manner described above by bradylobue.

PrusaSlicer
Version:   2.8.1+win64
Build:     PrusaSlicer-2.8.1+win64-202409181359

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 11 (build 26100), 64-bit edition
Total RAM size [MB]: 17,102MB
OpenGL installation
GL version:   4.6.0 (4.6.0 NVIDIA 560.94)
Profile:      Core
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce RTX 3060 Ti/PCIe/SSE2
GLSL version: 4.60.0
Textures compression:       Enabled

raven-drakeaurd, 2024-12-14:
Same issue here. Checking the console app, I get an HTTP error 500, followed by a 401, credentals not provided, then endless more HTTP error 500's until I kill the program. OP, do you have an Nvidia graphics card and update the drivers recently? It could be confirmation bias, or Nvida's market share, but it seems everyone I've seen with this issue has an GTX or RTX card.

akalderon, 2024-12-15:
Same issue here running on Mac OS Squoia 15.2 (24C101) - this is a bummer as I just finished setting up a new XL 5T and I can't get through PrusaSlicer to print any of my files.

mattsd88, 2024-12-15:
I am having the same issue.  Click the login button and I get a spinner and it says "(not responding)" in the upper left hand corner of the window.  

I was able to login a few days ago without any issue.  I updated my Nvidia drivers today and tried logging into the slicer after, so could be realted.  However, I did not methodically test logging in before and after the update so I cannot say this was 100% the issue. 

Prusa Slicer 2.8.1

OS: Windows 10 Pro 64-bit, Version 22H2
CPU: Intel Core i7-4770K CPU @3.5GHz
RAM: 16GB
GPU: EVGA GeForce RTX 3070 FTW3 ULTRA GAMING
GPU Driver:  Studio Driver 566.36 Tuesday December 10, 2024

Hope this helps, happy to provide more info, just let me know what you need.

Habakuck1956, 2024-12-15:
Same here, desktop PC Windows 11 24H2, slicer works on my laptop with older Windows 11 23H2. On my PC I recently updated the graphics card drivers. Did anybody contact Prusa on this issue yet?

MindlessDupe, 2024-12-15:
Yes. There's an active thread there also.

killwas, 2024-12-15:
Same issue on Vanilla OS (Linux).

Slic3r Version: PrusaSlicer 2.8.1 flatpak

The journal logs these errors:
# Error Log 

```
  19:56:46 slic3r_main: [2024-12-15 19:56:46.988025] [0x00007f5da0e006c0] [error]   HTTP Transient error (code=0, 
http_status=500), retrying in 64 s
  19:56:40 systemd: app-flatpak-com.prusa3d.PrusaSlicer-173285.scope: Consumed 5.375s CPU time, 1.1G memory peak.
  19:56:29 slic3r_main: [2024-12-15 19:56:29.517824] [0x00007f7be74006c0] [error]   HTTP Transient error (code=0, http_status=500), retrying in 64 s
  19:54:15 sh: Message: 19:54:15: Starting PrusaSlicer flatpak with entrypoint script
  19:54:15 systemd: Started app-flatpak-com.prusa3d.PrusaSlicer-173285.scope.
  19:54:15 systemd: Started app-flatpak-com.prusa3d.PrusaSlicer-173285.scope.
  19:54:15 systemd: Started app-gnome-com.prusa3d.PrusaSlicer-173285.scope - Application launched by gnome-shell.
```

This is my system:
# System Details Report
---

## Report details
- **Date generated:**                              2024-12-15 20:08:15

## Hardware Information:
- **Hardware Model:**                              Intel_R_ Client Systems NUC10i7FNH
- **Memory:**                                      64.0 GiB
- **Processor:**                                   Intel® Core™ i7-10710U × 12
- **Graphics:**                                    Intel® UHD Graphics (CML GT2)
- **Disk Capacity:**                               1.0 TB

## Software Information:
- **Firmware Version:**                            FNCML357.0052.2021.0409.1144
- **OS Name:**                                     Vanilla OS 2.0
- **OS Build:**                                    (null)
- **OS Type:**                                     64-bit
- **GNOME Version:**                               47
- **Windowing System:**                            Wayland
- **Kernel Version:**                              Linux 6.11.6-amd64

babylonman, 2024-12-15:
same on Mac.

Attempt to login either from first time config page or login button on main app UI.

MacBook Pro
Apple M1
Sequoia 15.0 (24A335)

kubispe1, 2024-12-16:
Hi,
On Thursday, we experienced unexpected downtime with PrusaAccount (https://github.com/prusa3d/PrusaSlicer/issues/13745). Although the system has since been relaunched, we are aware that some users are still encountering issues.

Our team is actively investigating the situation to determine the root cause and how to prevent such issues in the future. The department responsible for PrusaAccount is currently implementing changes to improve the situation.

Currently, some users have confirmed that they can sign in again. Please try as well and let us know.(https://github.com/prusa3d/PrusaSlicer/issues/13762).

MindlessDupe, 2024-12-17:
Hi there,
I am able to log in again to my account through PrusaSlicer - Connect tab and can execute the configuration wizard. Everything appears to be working properly. Thank you so much for your attention to this matter.
Wayne Louvier
Processor AMD Ryzen 5 5600X 6-Core Processor                3.70 GHzInstalled RAM 16.0 GBDevice ID 7474DF13-467F-4CCE-B173-C954BF7BE6E0Product ID 00325-81376-16197-AAOEMSystem type 64-bit operating system, x64-based processorPen and touch No pen or touch input is available for this display
PrusaSlicerVersion:   2.8.1+win64Build:     PrusaSlicer-2.8.1+win64-202409181359
Operating System:    WindowsSystem Architecture: 64 bitWindows Version:     Windows 11 (build 26100), 64-bit editionTotal RAM size [MB]: 17,102MBOpenGL installationGL version:   4.6.0 (4.6.0 NVIDIA 560.94)Profile:      CoreVendor:       NVIDIA CorporationRenderer:     NVIDIA GeForce RTX 3060 Ti/PCIe/SSE2GLSL version: 4.60.0Textures compression:       EnabledMultisampling: Enabled (4 samples)

Habakuck1956, 2024-12-17:
Everything seems to work just fine again … big thanks to the PRUSA team !!!Von meinem iPhone gesendet

closed 2026-09-29 by prusaslicer-bot[bot]