prusa3d/PrusaSlicer #13759: 2.9.0-beta1 for MacOS crashes when Send to Printer (via Prusa Connect) is selected

jack-21409, 2024-12-14:
### Description of the bug

2.9.0-beta1 crashes when I select Upload to Connect, for both MK3.9 and Mini+

### Project file & How to reproduce

How to reproduce: 

Slice a model
Select Send to Printer for a printer interfaced to Prusa Connect

MacOS Crash Log:

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               PrusaSlicer [1113]
Path:                  /Applications/Original Prusa Drivers/PrusaSlicer.app/Contents/MacOS/PrusaSlicer
Identifier:            com.prusa3d.slic3r/
Version:               PrusaSlicer PrusaSlicer-2.9.0-beta1 (PrusaSlicer-2.9.0-beta1)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-12-14 14:13:11.0482 -0500
OS Version:            macOS 15.1.1 (24B91)
Report Version:        12
Bridge OS Version:     9.1 (22P1072)
Anonymous UUID:        08FF916D-A600-8E56-8A22-E708E9E8C0F7

Sleep/Wake UUID:       F5008537-04BB-443E-A624-7E5BA91171BE

Time Awake Since Boot: 230000 seconds
Time Since Wake:       84274 seconds

System Integrity Protection: enabled

Crashed Thread:        0  slic3r_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 6 Abort trap: 6
Terminating Process:   PrusaSlicer [1113]

Application Specific Information:
abort() called

Thread 0 Crashed:: slic3r_main Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	    0x7ff810d0ab52 __pthread_kill + 10
1   libsystem_pthread.dylib       	    0x7ff810d44f85 pthread_kill + 262
2   libsystem_c.dylib             	    0x7ff810c65b19 abort + 126
3   libclang_rt.asan_osx_dynamic.dylib	       0x10b7e8556 __sanitizer::Abort() + 70
4   libclang_rt.asan_osx_dynamic.dylib	       0x10b7e7c84 __sanitizer::Die() + 196
5   libclang_rt.asan_osx_dynamic.dylib	       0x10b7cb197 __asan::ScopedInErrorReport::~ScopedInErrorReport() + 1207
6   libclang_rt.asan_osx_dynamic.dylib	       0x10b7ca437 __asan::ReportGenericError(unsigned long, unsigned long, unsigned long, unsigned long, bool, unsigned long, unsigned int, bool) + 1719
7   libclang_rt.asan_osx_dynamic.dylib	       0x10b7b5579 wrap_wcslen + 409
8   PrusaSlicer                   	       0x1054233cb Slic3r::GUI::PrintHostQueueDialog::on_progress(Slic3r::GUI::PrintHostQueueDialog::Event&) + 3083
9   PrusaSlicer                   	       0x10814d7dd wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 285
10  PrusaSlicer                   	       0x10814d50b wxEvtHandler::ProcessEventLocally(wxEvent&) + 59
11  PrusaSlicer                   	       0x10814d379 wxEvtHandler::ProcessEvent(wxEvent&) + 41
12  PrusaSlicer                   	       0x10814d009 wxEvtHandler::ProcessPendingEvents() + 313
13  PrusaSlicer                   	       0x1080813bf wxAppConsoleBase::ProcessPendingEvents() + 207
14  PrusaSlicer                   	       0x1081580a2 wxCFEventLoop::OSXCommonModeObserverCallBack(__CFRunLoopObserver*, int, void*) + 66
15  CoreFoundation                	    0x7ff810e29593 __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 23
16  CoreFoundation                	    0x7ff810e294ae __CFRunLoopDoObservers + 503
17  CoreFoundation                	    0x7ff810e28a47 __CFRunLoopRun + 905
18  CoreFoundation                	    0x7ff810e2808e CFRunLoopRunSpecific + 550
19  HIToolbox                     	    0x7ff81c541ec3 RunCurrentEventLoopInMode + 292
[... 470 lines cut ...]
USB Device: Touch Bar Display
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 47.5
Thunderbolt Bus: MacBook Pro, Apple Inc., 47.5

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0-beta1

### Operating system

MacOS 15.1.1 - Intel Mac

### Printer model

Prusa Mini and MK3.9

SachCZ, 2024-12-14:
Hello, this is a duplicate of #13748. There has been an error when releasing the application. Please download the dmg again. Sorry for the inconvenience.

closed 2024-12-14 by SachCZ

jack-21409, 2024-12-14:
Confirmed as working now in the latest DMG, thank you!