prusa3d/PrusaSlicer #13762: Unable to log into Prusa connect from Prusa Slicer

obi43, 2024-12-15:
### Description of the bug

Prusa slicer hangs when trying to "Log in". 

Problem exists on at least two different computers, in different physical locations. Note the hang is before user name is entered, and before any models loaded. Multiple restarts of slicer tried, even PC reboot.

Logging into Prusa connect from the mobile app works.

### Project file & How to reproduce

Not relevant. 

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer 2.8.1

### Operating system

Windows 10

### Printer model

Not relevant, crash/hang not dependent on printer

TheChucker98, 2024-12-16:
Also having this issue!

obi43, 2024-12-16:
Short update that might be of interest. 

It definitely seems like this is a log in issue.  On my work computer I were already logged into connect, and its still working. I guess if I log out I will no longer be able to log in, but for obvious reasons I do not want to trye this.  I have however on my work account tried to log into connect from a different computer, and then the slicer hangs. 

I do not exactly remember the last time I successfully logged into connect from the Slicer, but its maximum one week ago. The problem has in other words started sometime during the last week.

I have two different accounts on Prusa Connect, one for work and one for private, both have the same problem.

Logging into connect from the web site, or from the mobile works, only logging into connect from Prusa Slicer fails.

obi43, 2024-12-16:
It works now. Someone has done something, thanks :-)

TheChucker98, 2024-12-16:
WOOHOO! Thank you to whomever did whatever it was that you did! I can now print Christmas cookie cutters for friends in time for Christmas!!

kubispe1, 2024-12-16:
Hi,
On Thursday, we experienced unexpected downtime with PrusaAccount (#13745). Although the system has since been relaunched, we are aware that some users are still encountering issues.

Our team is actively investigating the situation to determine the root cause and how to prevent such issues in the future. The department responsible for PrusaAccount is currently implementing changes to improve the situation.

I’m glad to hear that it’s starting to work for you as well. Thank you for confirming!

closed 2024-12-17 by obi43