prusa3d/PrusaSlicer #13768: PrusaSlicer 2.8.1 - 2.9.0-beta1 will hang on Apple M1 - MacOS 15.2 (24C101)

elxris, 2024-12-16:
### Description of the bug

I was having issues to login to my prusa account, so I thought that it might be because uncompatible settings, so I erased everyting in `/Users/[username]/Library/ApplicationSupport/PrusaSlicer` and the beta folders. That only made things worse, I could not get past opening the app and the wizard because the application would hang. I downloaded 2.8.1 and the same. So I downloaded 2.7.5 and the application doesn't hang anymore.

I read that there was an old 2.9.0-beta1 dmg file but this is not the case, my dmg sha256 is `aef88582fd5414cc34650bd2efcc131346f39d7556cd9988d5b2b3b057d75eb4`

### Project file & How to reproduce

1. Download the  `PrusaSlicer-2.9.0-beta1.dmg` or `PrusaSlicer-2.8.1+macOS-universal-202409181403.dmg`
1.1 Clear all configuration files from 
2. Open the app
3. Expect the app to hang

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.1 to 2.9.0-beta1

### Operating system

Apple M1, macOS 15.2 (24C101)

### Printer model

MK4

elxris, 2024-12-16:
I have checked other recent issues, and all platforms report the same thing. I hope you can release a fix for this as it is outrageous that we cannot use the software if a server is causing all of these.

https://github.com/prusa3d/PrusaSlicer/issues/13765
https://github.com/prusa3d/PrusaSlicer/issues/13762
https://github.com/prusa3d/PrusaSlicer/issues/13758
https://github.com/prusa3d/PrusaSlicer/issues/13757
https://github.com/prusa3d/PrusaSlicer/issues/13748

kubispe1, 2024-12-16:
Hi, 
On Thursday, we experienced unexpected downtime with PrusaAccount. #13745 
Although the system has since been relaunched, we are aware that some users are still encountering issues.

Our team is actively investigating the situation to determine if the root cause lies on our side. The department responsible for PrusaAccount is currently implementing changes to improve the situation and ensure a smoother experience moving forward.

In the meantime, we kindly ask you to try logging in again and let us know if you still face any issues. Your feedback is invaluable and will help us address the problem more effectively.

We sincerely apologize for any inconvenience and thanks for cooperation

kubispe1, 2024-12-16:
Here, I note that some users are experiencing sign-in issues again (#13762).

elxris, 2024-12-16:
Yes, but thinking on many situations, like software archiving and offline setups, this represents an issue with the so promised offline use. The software should consider packing basic prusa profiles, even if they outdate quickly.

[image]

lukasmatena, 2024-12-17:
@elxris The software can still be used in offline mode, nothing changed about that. In fact, people who were offline were not affected by this problem.

There were two issues, one on the server side resulting in unexpected error, the other in PrusaSlicer (freeze when that unexpected error was returned). Both issues are now fixed, the fix in slicer is coming with the next release.

> The software should consider packing basic prusa profiles, even if they outdate quickly.

Up-to-date profiles (not just Prusa profiles, but also the 3rd party ones) are bundled with every release. Actually, they always have been.

elxris, 2024-12-18:
Thank you for the detailed response, I'll proceed to close this.

closed 2024-12-18 by elxris