prusa3d/PrusaSlicer #13773: Prusa Slicer 2.9.0-beta1 crashing if using "Send to Connect"

Graloth, 2024-12-16:
### Description of the bug

I wanted to give the improved multi-bed slicing a try, but it seems that when trying to upload any sliced file to the printer via the "Send to Connect", instantly crashes the app.

The only way for it not to crash is to save the generated gcode locally and then manually uploading it to Prusa Connect via the browser.

### Project file & How to reproduce

1. Open Prusa Slicer 2.9.0-beta1
2. Log in to Prusa Connect inside the slicer, if you haven't already
3. Add _any_ model to be sliced, in my tests I just added a basic cube shape
4. Slice the model
5. Click the "Send to Connect"
6. See that the application instantly crashes, triggering the macOS crash reporter.

Below are linked the crash report generated by macOS, as well as a video showing the exact steps taken (as listed above) to trigger the crash.

- Crash report: [macos-crash-report.txt](https://github.com/user-attachments/files/18147864/macos-crash-report.txt)
- Video: https://youtu.be/LXA3MeKkPiU

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0-beta1

### Operating system

macOS Sequoia 15.1.1

### Printer model

Prusa MK4S

SachCZ, 2024-12-16:
Hello, sorry there has been a mistake when releasing the macOS beta version. Please refer to https://github.com/prusa3d/PrusaSlicer/issues/13750#issuecomment-2543089390

It should fix your issue. Thank you for your patience.

Graloth, 2024-12-16:
> Hello, sorry there has been a mistake when releasing the macOS beta version. Please refer to [#13750 (comment)](https://github.com/prusa3d/PrusaSlicer/issues/13750#issuecomment-2543089390)
> 
> It should fix your issue. Thank you for your patience.

Thanks, that did indeed fix it!

closed 2024-12-16 by Graloth