prusa3d/PrusaSlicer #13777: Slice button disabled when adding shape from gallery

valterfc, 2024-12-16:
### Description of the bug

In a new project, when trying to add a shape from the gallery, the slice button is not activated.

[image]

[image]

### Project file & How to reproduce

New project.

On the plate, add shape from the gallery.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.9.0 beta1

### Operating system

Windows 10

### Printer model

Custom

lukasmatena, 2024-12-17:
Thanks for the report. This is a known issue, it will be fixed in 2.9.0-rc1.

TerPuc, 2024-12-17:
Issue created: SPE 2617

lukasmatena, 2024-12-18:
Fixed in 2.9.0-rc1. Closing.