prusa3d/PrusaSlicer #13778: No Dark Color Scheme

AeroSonic4, 2024-12-16:
### Description of the bug

The move to Flatpak the system color scheme is not picked up. I use a dark system color scheme.  It is picked up on previous AppImage, but not on Flatpak.  

### Project file & How to reproduce

Open the program and the dark configuration does not come up. 

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.9.0-beta1

### Operating system

KDE Neon

### Printer model

Ender 3 Pro with BT extruder

Jan-Soustruznik, 2024-12-17:
Hello, @AeroSonic4 ,
Did you encounter the same issue with the Alpha1 release of PrusaSlicer 2.9.0? Could you try restarting your PC after installing PrusaSlicer?

I’m using Ubuntu 22 and I haven’t this problemCould you provide more details about your OS? Which GUI are you using—default or custom?

Thank you!

AeroSonic4, 2024-12-17:
Hi @Jan-Soustruznik 

Here is information about my system:

Operating System: KDE neon 6.2
KDE Plasma Version: 6.2.0
KDE Frameworks Version: 6.6.0
Qt Version: 6.7.2
Kernel Version: 6.8.0-49-generic (64-bit)
Graphics Platform: X11
Processors: 8 × Intel® Core™ i7-6820HQ CPU @ 2.70GHz
Memory: 15.3 GiB of RAM
Graphics Processor: Mesa Intel® HD Graphics 530
Manufacturer: Dell Inc.
Product Name: Precision 7510

My KDE Neon is using Breeze Dark as the Global Theme. 

I'm that familar with Flathub/Flatpak and don't see any place to switch to the alpha version.  If I did, I would give it a try.  I prefer AppImage as I have more control of what I am using and can have multiple versions installed.    

Also, Here is what I have under the GUI tab in PrusaSlicer:

[image]

I hope this helps, If you have any more questions please ask.

SachCZ, 2024-12-18:
Hello @AeroSonic4, while we work on the issue reproduction you may use the following workaround to run PrusaSlicer with dark theme from the command line. 

```
flatpak run --env=PRUSA_SLICER_DARK_THEME=true com.prusa3d.PrusaSlicer//beta # if you have the flathub beta repository available and want to use the latest pre-release build
flatpak run --env=PRUSA_SLICER_DARK_THEME=true com.prusa3d.PrusaSlicer//stable # for the latest stable version
```
We are currently working with the version that the dark mode is not correctly detected on your system, but if the mentioned commands do not work for you it might be a different issue.

Sorry for the inconvenience and if you could please let us know about your findings as it might help resolving the issue.

SachCZ, 2024-12-18:
Also a followup. If you prefer a more user friendly and permanent solution to set the env variable, you can use [flatsteal](https://flathub.org/apps/com.github.tchx84.Flatseal). You can install it just as you install any other flatpak.

```
flatpak install flathub com.github.tchx84.Flatseal
```
After that to change the ENV variable do:
```
flatpak run com.github.tchx84.Flatseal # or run it from the GUI
```
And go to `PrusaSlicer -> Environment` and set:
```
PRUSA_SLICER_DARK_THEME=true
```

Hope this helps.

AeroSonic4, 2024-12-18:
Thank-you.

Yes, it works by installing Flatseal and running it.  Next going to the Prusaslicer application setting in Flatseal and changing the Environment Variable to true.   Here is what it looks like:

 
[image]

SachCZ, 2024-12-19:
Thank you for the info. This is very helpful.

NHodgesVFX, 2024-12-20:
Just wanted to report that dark mode does not work on the Flatpak version out of the box for me either. I can confirm setting the environment variable PRUSA_SLICER_DARK_THEME=true in Flatseal for PrusaSlicer fixes the issue. Interestingly, I noticed the environment variable PRUSA_SLICER_DARK_THEME was already there and set to false. So I just changed it to true.

PrusaSlicer
Version:   2.9.0
Build:     PrusaSlicer-2.9.0+flathub.org

Operating System: Kubuntu 24.04
KDE Plasma Version: 5.27.11
KDE Frameworks Version: 5.115.0
Qt Version: 5.15.13
Kernel Version: 6.8.0-50-generic (64-bit)
Graphics Platform: X11
Processors: 32 × AMD Ryzen 9 7950X 16-Core Processor
Memory: 30.5 GiB of RAM
Graphics Processor: NVIDIA GeForce RTX 2080 SUPER/PCIe/SSE2
Manufacturer: ASRock
Product Name: X670E Pro RS

tobymurray, 2024-12-27:
Maybe not helpful, but I saw the same outcome (dark mode didn't work) moving from 2.8.1 to 2.9.0 (not beta) installed via flatpak.

Operating System: EndeavourOS 2024.06.15
Kernel Version: 6.12.6-arch1-1

etc.

Lux3, 2024-12-30:
> Also of note someone posted in the comments on [this article](https://help.prusa3d.com/article/dark-mode_258977) that they were having the same issue as well.
>

I'm the one who commented on the knowledgebase.
No dark mode option in 2.9 in alpha, beta, or release since switching to flatpak.
OS: Garuda Dragonized Gaming Edition (Arch Linux)
Kernel: 6.12.6-zen1-1-zen x86_64

Same issue on Debian 12 and TuxedoOS using flatpak. I've had darkmode set on dozens of system themes in an attempt to make it work. Trying the env variable for dark mode, which does work for now.

Jan-Soustruznik, 2025-01-23:
Hello, @AeroSonic4, thank you for your report. We will try to fix in some of new release, issue created(SPE-2655)

bigblzrd, 2025-01-24:
> Also a followup. If you prefer a more user friendly and permanent solution to set the env variable, you can use [flatsteal](https://flathub.org/apps/com.github.tchx84.Flatseal). You can install it [...]

HI there, I came searching for a fix to this issue and this worked brilliantly for me! It also led me to to play with flatseal a bit and I found a way to make the theming consistent for me on KDE. This solution was found on the EndeavourOS forum.

Go into flatseal
Go to flatseal>all applications>Filesystem>Other files
Add variable and enter~/.themes in field
Go to flatseal>all applications>Filesystem>Environment>Variables
Add variable and enter the name of your theme e.g. GTK_THEME=Breeze

I'm now able to use my main Breeze theme so the application matches my system wide theme. Just make sure to go back to PrusaSlicer (BOSS) and remove the 'PRUSA_SLICER_DARK_THEME=true' variable and leave only the 'GTK_THEME=Breeze' variable.

I hope this helps someone :)

closed 2026-09-29 by prusaslicer-bot[bot]