prusa3d/PrusaSlicer #13779: 2.9.0-beta1 Failes to generate infill on painted multicolor instance

rcmaniac25, 2024-12-17:
### Description of the bug

I painted a model, enabled wipe into infill. Then to try to reduce purge tower, I created an additional instance. The wipe tower didn't change. I went to look at the infill and the 2nd instance of the model was missing most of it's infill.

### Project file & How to reproduce

[N64.zip](https://github.com/user-attachments/files/18162143/N64.zip)

1. Open the project
2. Slice it and check for infill (it should be there)
3. Select the N and click + on the keyboard or the options bar above
4. Move the new instance to a different location and slice.
5. The infill of the 2nd instance should be mostly missing.

1.
[image]
2.
[image]
3 & 4.
[image]
5.
[image]
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.9.0-beta1

### Operating system

Windows 10

### Printer model

Prusa MK4S+MMU3

Jan-Soustruznik, 2024-12-17:
Hello @rcmaniac25,Thank you for your report. We can confirm that there is issue, but this problem can also be reproduced in older versions.The issue is most likely caused by the additional option "Wipe into this object infill". When you slice the model a second time, the infill isn’t re-sliced; instead, only the G-code is re-generated. As a result, the wipe (infill) is missing for instance.

internal issue created: SPE-2626

lukasmatena, 2025-02-18:
Should be fixed in [PrusaSlicer 2.9.1-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.1-alpha1). Closing.

closed 2025-02-18 by lukasmatena