prusa3d/PrusaSlicer #13781: 2.9.0-beta1 Crashes when sending to connect

Zendemic, 2024-12-17:
### Description of the bug

Prusa Connect constantly crashes whenever I click "send to connect". Despite the crash, it appears the GCODE file is still successfully sent and begins printing normally 

### Project file & How to reproduce

[isometric-grid-first-layer-020-a7-74x105.3mf.zip](https://github.com/user-attachments/files/18169289/isometric-grid-first-layer-020-a7-74x105.3mf.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.9-beta1

### Operating system

Mac OS 15.0 Beta on a 2019 Macbook Pro 16.1" (intel)

### Printer model

Prusa Mini

lukasmatena, 2024-12-17:
Thanks. It can be caused by https://github.com/prusa3d/PrusaSlicer/issues/13750#issuecomment-2543089390. Can you check that?

Zendemic, 2024-12-17:
That appears to have done it! I wasn't even aware of the issue in the first place. Thanks for the help!

closed 2024-12-17 by Zendemic