prusa3d/PrusaSlicer #13782: Differentiate internal and external bridge for separate control

5jvm0u4, 2024-12-17:
**Is your feature request related to a problem? Please describe.**
For certain high temp material, due to insufficient cooling while bridging INTERNALLY to form the top layer, the top layer will sag, but when it's EXTERNAL bridging there's plenty of room for air to flow through, thus bridging without sagging.

**Describe the solution you'd like**
I'd like the slicer be able to separate parameter related to bridging into external and internal, like orcaslicer does, including Bridge_to_flow_ratio, Bridge_speed, Bridge_fan_speed and also offer this in filament overwrite section.

**Describe alternatives you've considered**
Or maybe consider changing infill from Grid to Rectilinear for several layers before internal bridging.

**Additional context**
The photo shows Bridge_to_flow_ratio set to 0.8, 0.9, 1, both 0.9 and 1 is touching print bed, which means it had sagged more than 1mm
[image]

5jvm0u4, 2025-03-02:
OrcaSlicer now has this featuer along with a lot more
https://github.com/SoftFever/OrcaSlicer/pull/8263

shinanca, 2026-09-14:
It`s really need!