prusa3d/PrusaSlicer #13784: Wrong calculation of 'minimal filament extrusion length' causing very wide skirt

AmadeusW, 2024-12-18:
### Description of the bug

Using default settings of "minimal filament extrusion length"=4mm, the skirt around each object is very wide (~25mm). This is **not** a brim. Brim is off. This is a three layer disc, an unnecessarily large skirt.
[image]

When I started using slicer a few years ago, the skirt was a single loop. Now the skirt is a very large object with the default of 4mm Looks like it's a bug - additional 4956mm^3 of filament are used because of this setting at 4.

Here are screenshots showing values of 0 (how I know skirt when started using PrusaSlicer a few years ago), 1 and 4 (the default)
- 0 (what I expect to be the correct setting)
[image]
- 1 (for comparison)
[image]
- 4 (the new default since I updated a few months ago)
[image]

### Project file & How to reproduce

[skirt bug repro.zip](https://github.com/user-attachments/files/18175506/skirt.bug.repro.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.1

### Operating system

Windows 11

### Printer model

MK3S+

rtyr, 2024-12-18:
The issue is caused by your custom filament profile, where the diameter of the filament is set to 22.99mm.

closed 2024-12-18 by rtyr

AmadeusW, 2024-12-20:
Thank you for looking into it, this addressed a slew of extrusion issues that started happening recently.