prusa3d/PrusaSlicer #13785: Modifier "Ironing - only on top" not applied

Moyrn, 2024-12-18:
### Description of the bug

I want to modify a surface to only iron the top surface - using a modifier
It is not applied
See pics
[image]

if i select "all layers" or "all top layers" it works
[image]

### Project file & How to reproduce

[SEITA-IRONING-BUG.zip](https://github.com/user-attachments/files/18176365/SEITA-IRONING-BUG.zip)

i added a simple modifier to play with ironing option : simply enable, and try "only top" vs other options

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.1

### Operating system

Win11

### Printer model

MK4, XL, MINI

Moyrn, 2024-12-18:
same behaviour on  2.9.1

u89djt, 2024-12-18:
(Fellow user taking an interest) It seems to working correctly in your example:
[image]

Moyrn, 2024-12-18:
you are right. my object was too big... so i cut it in two... mistakenly removing the higher ground
so in that context, my uploaded project was not representative

i just re-did it with a simple cube & cylinder game

many thanks !

u89djt, 2024-12-18:
Did you update both copies of the file? They have different names now... We can delete this new comment and your response to keep the noise down

u89djt, 2024-12-18:
Actually once we've made sure the example works, we can delete all the comments since they don't serve a purpose to the repository

u89djt, 2024-12-18:
Yeah, the second instance of a project file in your post is still the old one. 
[image]

u89djt, 2024-12-18:
I think "only top surface" means the very top of the object, so it's doing what you asked. It seems you have to a little bit more work crafting the modifier. Here's a playground with a couple more features to see what's happening.
[Uploading ironingsteps.zip…]()
[image]

u89djt, 2024-12-18:
I think your understanding of it would be a better way for it to function, and hence a natural assumption, but I'm sure it isn't how it functions currently.

closed 2026-09-29 by prusaslicer-bot[bot]