prusa3d/PrusaSlicer #13786: slicer err for sl1 printer

xingshh, 2024-12-18:
### Description of the bug

slicer error for model with cavity that has indepented mesh . 
[ball.zip](https://github.com/user-attachments/files/18178267/ball.zip)

### Project file & How to reproduce

1, select sl1 printer 
2, add model 
3, slice 

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.1

### Operating system

window10

### Printer model

prusla sl1

xingshh, 2024-12-18:
[image]

Jony01, 2024-12-19:
I confirm, that issue with inner shell is reproducible and will be fixed in future.
At this moment, as a way around, you could split your model `To parts`.
[image]
And select inner shell as negative volume.
[image]
[image]
And you will achiev your desired model
[image]
Do not forget to add drain holes to not print closed reservoar of the resin.

xingshh, 2025-01-07:
@Jony01  
i read the code in file SLAPrintSteps.cpp,  in function 
void SLAPrint::Steps::mesh_assembly(SLAPrintObject &po)
{
    po.m_mesh_to_slice.clear();
    po.m_supportdata.reset();
    po.m_hollowing_data.reset();

    csg::model_to_csgmesh(*po.model_object(), po.trafo(),
                          csg_inserter{po.m_mesh_to_slice, slaposAssembly},
                          csg::mpartsPositive | csg::mpartsNegative | csg::mpartsDoSplits);

    generate_preview(po, slaposAssembly);
}
it is caused by flag csg::mpartsDoSplits. Can this flag be removed？

Jony01, 2025-01-08:
It is not my code. So, I barely see impact of the change.

By [commit](https://github.com/prusa3d/PrusaSlicer/commit/25ca46e3eb14b2349fba7b5c9980f99de94fe9ec)  it was added to support for csg operation stacking by @tamasmeszaros

@tamasmeszaros : Could you point as what will not work after removing the flag?

I gave it a try and It seems that this would probably fix multi shell model input stl file usage.

closed 2025-02-18 by lukasmatena