prusa3d/PrusaSlicer #13787: Crashes in 2.9.0 Beta

3DThangs, 2024-12-18:
### Description of the bug

I was getting ready to upload my gcode to my printer using the "Send to Connect" button, but it crashes when I do. I repeated this, and it happened once more.

https://github.com/user-attachments/assets/ae3e68d6-282f-48f2-ac16-046130f544af

### Project file & How to reproduce

[All.3mf.zip](https://github.com/user-attachments/files/18181166/All.3mf.zip)
When opening this in the latest beta, just export it to connect, and then it will crash.

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0 Beta1

### Operating system

MacOS Sequoia

### Printer model

Prusa Mini+

TerPuc, 2024-12-18:
Dear @3DThangs, thank you for your message. Could you please try what my colleague described in this [comment ](https://github.com/prusa3d/PrusaSlicer/issues/13750#issuecomment-2543089390 ) and let us know if the crash is still happening? Thanks!

3DThangs, 2024-12-18:
> Dear @3DThangs, thank you for your message. Could you please try what my colleague described in this [comment ](https://github.com/prusa3d/PrusaSlicer/issues/13750#issuecomment-2543089390) and let us know if the crash is still happening? Thanks!

Thank you for your quick reply. The issue is fixed in the new [2.9.0 rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-rc1) version that I just downloaded. Thanks.

closed 2024-12-18 by lukasmatena