prusa3d/PrusaSlicer #13791: [2.9.0-rc1] Custom bed model / texture not showing for multi-plate

mjonuschat, 2024-12-18:
### Description of the bug

Bug: Having a printer defined with a custom bed texture and a custom bed model leads to neither the texture nor the model being show when a second plate is added. When only using a bed texture these show up when using more than one plate.

[image]

Expected: Every plate shows custom bed texture and model

### Project file & How to reproduce

1. Define a printer that has a custom bed texture and model:
   [image]

2. Add two shapes to the plate - texture and model are still visible
   [image]

3. Move one shape to a second plate - model and texture vanish (as shown in screenshot in the description of the bug)

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.9.0-rc1

### Operating system

MacOS 15.1.1-arm64

### Printer model

Voron Trident 150mm

lukasmatena, 2024-12-19:
Thanks. This is actually by design. Your bed model is so large that the beds would have to be spaced awkwardly apart to avoid overlaps. In such case it falls back to the default rendering. If the model was smaller, it would keep showing even with multiple beds.

What does look suspicious is that neither of the beds is rendered as active in your screenshot. That seems to be a bug. We will try to reproduce it.

lukasmatena, 2024-12-20:
The issue should be fixed in [2.9.0](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0). Closing.

closed 2024-12-20 by lukasmatena